polyend/TrackerBetaTesting #709: [Firmware version] Bug title

sunfallz, 2021-01-17:
- Midi CC51 is for slot one in Performance mode, i set it to Pattern Length 
- external cc values only work in performance mode
- i could only get the Pattern Length external modulation to kind of move in increments of 1, 2, 4, 8, 16. 

nothing in between and nothing over, and it was extremely glitchy at best. not really holding on any one value. 
so not optimal just yet. hopefully an update will allow for any length modulation from 1-128
steps within a pattern

* How often does this bug happen? - every time,

* Does this bug happen in previous firmware versions? untested in previous firmware

* Does this bug happen in every project? Yes

piotr3500, 2021-02-03:
The values fo Pattern Length are fixed to values: 1, 2, 4, 8, 16.
It has nothing to do with CC messages.

closed 2021-02-03 by piotr3500

sunfallz, 2021-02-03:
> The values fo Pattern Length are fixed to values: 1, 2, 4, 8, 16.
> It has nothing to do with CC messages.

Thanks for the clarification. So CC values we are to enter for Pattern Length are locked at 1, 2, 4, 8, 16? 
That is odd to say the least. Reckon this should've been listed as a feature request.
Cheers