polyend/TrackerBetaTesting #710: Hit button to walk through pattern line for line WITH sound

endrabergon, 2021-01-18:
This feature was present in some older trackers like Protracker. Every time you pressed the Return button it would play through all notes in that specific line number of the pattern. Very handy for auditioning what notes you have on each line, as slow or fast as you wanted. Would have loved it to be implemented!

ConnorCarbon, 2021-01-27:
Yes, like this option idea a lot. As a new user I have found myself wondering if there is a happy medium between "Shift+Play" to get an audible sense of where I am in a pattern, and pressing "Preview" on a given step to preview only that one track's single step, which is a two-handed and manual operation.
Having a "timeline scrub" equivalent on the Tracker would be valuable for maintaining a referential sense of where in time the user is. This is a standard feature of video editing software, computer DAWs, and so on and can really accelerate workflow but also bring the user closer into the "soundcrafting space"

endrabergon, 2021-01-27:
Yeah, its a great way to audition what you have in the pattern. In Protracker it was called the STEP button (Return key on the keyboard).