polyend/TrackerBetaTesting #711: Software midi thru

ROM72, 2021-01-18:
I searched on both fb groups and on the github page and I haven't found anything about it (except my own post and a quick reply from a Polyend employee saying simply PT doesn't do it)

It will be nice to have software midi thru on PT.
It allows more midi routing possibilities, it's very convenient.
For example, I want to send a midi lfo from my digitone to that same digitone to have more changing values. 

I can't for now, the PT is my only machine (I own a keystep, a Digitone, a Machinedrum, a Shruthi) which can't pass thru midi cc.

I don't know if hardware wise the tracker would be able to do so or not, but it seems like a basic feature.

UserOverflow, 2021-01-22:
I just wanted to report it, but I'm glad that this thread has already been created.
For a portable setup, this is a very important option, for example, I use this in Modal Skulpt, so I can connect devices in series (no splitters required). In addition, using one midi keyboard for the tracker, you could also control it on other devices (on other channels).

ROM72, 2021-01-22:
Yes you're right, I haven't thought about this use case, this is the type of setup I could have in the future though.

peblin, 2021-03-09:
I really agree with this request - I think there should be a setting to pass thru any midi received except for the channel set for the PT "Notes in channel".

It would be super-convenient and allow daisy-chaining of Midi controller -> PT -> outboard synths.

OmskInfo, 2021-04-10:
+1000 as well. I would love a simple "MIDI Thru" mode for the MIDI Out jack that just passes everything 1:1 from the input jack to the output jack—I wouldn't need any additional output from the Tracker to be merged into the thru for my particular use-case.

ROM72, 2021-07-06:
Bumps, we really need this simple feature, 
I need to pass thru my midi keyboard notes to multiple synths including PT.

bestenfallskongenial, 2021-09-28:
I ask for it since I got mine, summer 2020.
I think it’s a very important feature and ask myself why nothing happened jet