polyend/TrackerBetaTesting #712: [1.3.0] High pitched noise in all audio output

mfroach, 2021-01-18:
> After updating to 1.3.0 firmware, at all times there is a subtle high
> pitch noise from audio output. When line in monitor is on it is significantly
> louder, and master volume affects the volume of the noise.

PEtajaja, 2021-01-18:
Out of curiosity, does the noise stop when you unplug everything?

mfroach, 2021-01-18:
When removing the line out , yes the audio stops playing and I cannot hear the noise anymore

PEtajaja, 2021-01-18:
Should've been clearer, remove everything and listen via headphones rather than through an interface / mixer etc.

I've had similar issues with gear before that ended up being the result of a ground loop or usb noise, if it was to go away by doing the above, could it be you changed anything else?

Obviously if the issue stays then it's caused by the update

doelman, 2021-01-18:
For me since 1.3 the Tracker actually outputs a buzzing sound when it's switched off (but plugged in, obviously). Dissapears as soon as I switch the unit on

mfroach, 2021-01-18:
Ah, yes, the issue remains with direct headphone connection. I have also tried swapping 3.5 cable  resetting firmware (1.3 again), with no luck. This didn’t happen before firmware update. Also, the noise only starts once the device had played some audio(playing a pattern or sample)

mfroach, 2021-01-19:
Am curious if this is related to this new feature;
> Live Audio Line Input Monitoring is now available during song playback

mfroach, 2021-01-19:
Muting all tracks and line in gets rids of the noise. So not a hardware issue I think.

ambv, 2021-01-19:
Does muting Line In only get rid of the noise as well? Not sure how muting other tracks can affect noise when not playing anything.

mfroach, 2021-01-19:
Muting line in gets rid of some of the noise , muting the tracks with notes on them gets rid of all the noise. Also for clarification I’ve managed to make this barely noticeable by setting master volume to ~25db and turning up volume for dry mix

t8pt, 2021-01-19:
Had this problem with constant background noise after recording from line in, too.
Newly created projects suffer from this background noise. When opening one of the pre-made projects made by other artists the static noise is not present. Very strange is that if I change the display brightness to max the noise is gone, too. After fooling around a bit I found a temporary workaround: I reverted the firmware to 1.2.0, created a init project (without any background noise), updated to 1.3.0 and use this init project as a starting point.

t8pt, 2021-01-19:
To be honest I'm questioning if this static noise from the line in is solely software dependent. Could there be some kind of hardware problem?

t8pt, 2021-01-19:
Checked again - line in noise is severely reduced after changing the display brightness from low to high. Not good because this feels like it's an electronic hardware based problem...

anemlife, 2021-01-20:
> Checked again - line in noise is severely reduced after changing the display brightness from low to high. Not good because this feels like it's an electronic hardware based problem...

Maybe not related but I had some similar issues with my modular system when plugged at home (old house, bad installation). Got rid of all the digital noises by using a D.I. box. In my current setup the tracker output is plugged to the modular then to the D.I. box then to my soundcard and it feels ok about the noises! Maybe try a D.I. box? 

Also I always keep the tracker master level between -20db and 0db and then I boost it with external gear because when I go up to 0db internally I do notice some noise! But boosting it after sounds better in my opinion.

Just curious but what are you using to power the Tracker?

t8pt, 2021-01-20:
> Just curious but what are you using to power the Tracker?

Wall mount and USB power bank. The issue occurs with both and at different places (old house, new house, etc.).

tinysynths, 2021-01-20:
Same issue here. The Tracker line out is nearly completely silent (in fact better signal to noise than many products - it is great). However, when line in is active both for monitoring and when recording samples there is a very annoying high pitched noise. I've tested without anything plugged into the line in and with various synths/instruments plugged in. The high pitched noise gets recorded to the signal and makes the sounds unusable. Pitch shifting a sample recorded via the line in makes the noise even more present.

I have run various tests and it does not seem to be a ground loop issue. I have run similar tests with my eurorack, Elektron devices, microfreak, outboard gear, guitar pedals etc. Only the Tracker exhibits this noise (when line in is active).

I have tested powering the Tracker using battery packs, usb hub, usb c port on laptop and wall adaptor. The noise issue doesn't change or get affected by different power sources and is always present.

Attached is a screenshot of the noise.

[image]

It would be great to get this issue fixed so we can record noise free samples.

Thanks!

t8pt, 2021-01-21:
> Same issue here.

Do different settings of display brightness change anything?

mfroach, 2021-01-21:
> Very strange is that if I change the display brightness to max the noise is gone, too

can confirm on my machine this lowers the volume of the noise (does not get rid of it completely.)on low brightness it’s much louder. And fun fact, my bluetooth speaker suddenly gets the same noise when I change brightness to low. I don’t hook the speaker up to the tracker. Seems like the display is causing some kind of RF noise??

CV-Gate, 2021-01-22:
I never had this issue but, in my case, I always power from a power bank. Maybe related?

thebradclay, 2021-01-22:
@CV-Gate  it has been mentioned by @t8pt that the issue persists even with a power bank,
"Wall mount and USB **power bank**. The issue occurs with both and at different places (old house, new house, etc.).".

topoexplo, 2021-01-29:
I’m also experiencing this issue. The noise is high pitched and wavy. And yes, the issue is not as bad when I change the screen brightness to high.

gurkenkimchi, 2021-01-29:
i can confirm this issue too. is there going to be a fix for it or its mainly a hardware issue ?

t8pt, 2021-01-29:
> is there going to be a fix for it or its mainly a hardware issue ?

the question!

t8pt, 2021-02-20:
the question still remains...

mfroach, 2021-02-20:
> the question still remains...

@t8pt to be fair the issue is possibly unfixable and pretty negligible. since it seems like the display is causing the noise it could require a hardware fix. Plus i don’t notice it at all once lowering master volume. would prefer it gone but I’ll be patient

ambv, 2021-02-28:
This is indeed display brightness related and is fixed in 1.3.1 beta 4.

closed 2021-02-28 by ambv

demolen, 2021-12-03:
This is not fixed in 1.4.0 or at least regressed.