polyend/TrackerBetaTesting #713: Midi Program Change In

talktogreg, 2021-01-19:
Ability to configure two inbound MIDI channels to accept Program Change messages

1) Select a channel (Off, 1-16, ALL) to choose current Instrument (1-48 + 59-64[midi])
2)Select a channel (Off, 1-16, ALL) to enable Pattern selection (first 128 patterns due to MIDI index limitations)

First feature would be very useful when using an external MIDI controller keyboard, second great for controlling Tracker via an external sequencer.

tc-joegiralt, 2021-04-05:
the second feature would be amazing :)

+1 Select a channel (Off, 1-16, ALL) to enable Pattern selection (first 128 patterns due to MIDI index limitations)

DistantAnimals, 2021-06-20:
selecting pattern via program change (or perhaps even by using the main tracker buttons whilst in Song mode) would be my number one feature request.

PEtajaja, 2021-12-05:
I requested this a while back and I know a few others were really keen on it too.. If the pattern select via program change could happen instantly or at least have the option to, that'd be amazing and would be a pretty huge deal for me, it allows you to shuffle patterns together as well as being able to arrange patterns in interesting ways via an external sequencer rather than the inbuilt song mode.

PEtajaja, 2022-01-23:
So happy to see this has made its ways onto the milestones for 1.6! I've been on and off considering selling my tracker for a while as it didn't have its own place amongst the rest of my gear but if midi controlled instant pattern changes become possible in such a way that we can shuffle patterns together by rapidly changing back and forth with prog change and / or cc messages, this would 100% be reason for me to keep mine to use together with other gear that's capable of this.

PEtajaja, 2022-01-28:
I've been thinking.. while it'd be great to be able to do the above.. I still think the most useful would be if we were able to trigger an instant pattern change via midi through the performance section. There's some very useful functions in the performance section that'd be very useful when using an external sequencer as our song mode.

ROM72, 2022-07-01:
Well not in 1.6 but I hope it will be here for the next update <3