polyend/TrackerBetaTesting #716: [1.3.0] Pattern button + Arrow = Delete?

thebradclay, 2021-01-20:
## Hi, sorry you had trouble!

### Steps to reproduce

Over the last few days I've noticed that occasionally while listening to patterns play when I press the pattern button (the one on the right, beside fx2) and an arrow key sometimes my pattern will delete. This has happened just now but I can't consistently reproduce it. It seems to happen if I press the  button combo right as the pattern is ending.

### Attachments and links

### Other important information

* How often does this bug happen? Seems random but more likely if the buttons are pressed as the pattern is ending

* Does this bug happen in previous firmware versions? Not that I am aware of

* Does this bug happen in every project? Has happened in three so far.

piotr3500, 2021-01-20:
Are you sure that this combination doesn't change the pattern (from one to another)?

thebradclay, 2021-01-20:
@piotr3500 It does change the pattern. What I am saying is that it has also deleted the entire pattern occasionally that is why the title has a question mark. As in why is this happening.  So to clarify yes it does change the pattern, which it should do. However, this same combination has also resulted in deletion of the pattern which it should not.

piotr3500, 2021-01-20:
Got it, thanks for clarifying.
We'll look into it.

ambv, 2021-01-24:
Repro: https://www.dropbox.com/s/yuum0lduu4vtcxg/PatternDeletedByWellTimedSwitch.mov?dl=0

ambv, 2021-08-20:
An easier repro is that you can confuse the sequencer by letting go "sync pattern change" on last step of pattern, current pattern number doesn't get updated but the new pattern gets loaded.

closed 2021-08-20 by ambv