polyend/TrackerBetaTesting #719: Suggestion --> Option to enter MIDI instruments with pads

anemlife, 2021-01-20:
When you write a pattern (instrument mode), if you want to write different notes from different instruments you can use the pads. But this only works for instrument 1-48 and not with MIDI channels! 

I understand that this is due to the limitation of 48 instruments/pads, but what if the pads (when entering instruments) could be related to the last type of instrument entered, so we could also switch between midi channels fasters!

Example: 

1) I add a note for instrument 1 => all the pads are now displaying instruments from tracker
2) I add a note for MIDI Channel 1 => all the pads are now displaying MIDI channels

Let me know if anything sounds wrong!

eholzbach, 2021-01-24:
I'd vote for a shortcut. Something like shift + instrument + pad 1 = midi channel 1.

anemlife, 2021-01-24:
> I'd vote for a shortcut. Something like shift + instrument + pad 1 = midi channel 1.

Nice idea! This could perfectly do the trick :)

closed 2021-12-02 by Jacknametrouble