polyend/TrackerBetaTesting #721: [1.3] Program change not in perfect timming, missing 1st step

Tombelliard, 2021-01-21:
## Hi, sorry you had trouble!
No problem, Im very happy to be part of the tracker community and Im shure that the tracker will be better and better day after day. 

### Steps to reproduce

Im sending program change to a drum machine (tr8s) at the first step of a pattern in tracker.  
Each time the pattern come to step one, in the tracker, with the fx Prg change on it.... the first step and note, (on a the tr8), (usualy kick) look like to have a big attack up. We only hear a little "ummm"... and unfortunately its doesnt have the same effect ! 
May be the problem can be fixe if micro timming can be set in negative cause its look like a latency problem. I dont know.
Im shure in my midi chaining.. Tracker>midi thru box>tr8 

Some one get "kind of" same problem of program change, in issue #391 (with other latency level, much more). But it was version 1.2.2 and an other user ask him to try with the newest beta version if problem occurs. 
I would love to try beta version. But does it possible to come back to official version if to many bug come? 

### Attachments and links

I try to do my best
https://www.youtube.com/watch?v=auvg0jg-ANo 

### Other important information

* How often does this bug happen? (every time, often, "it only happened once")

I start to try prog change yesterday and from yesterday, yes... everytime.
Yesterday I was thinking to put the program change on the last step of the pattern to avoid the problem.
But if the pattern is looping, its doesnt work...
That why micro timming in negative can maybe a good option if its possible

* Does this bug happen in previous firmware versions?

 I dont know... I use it since two week only, but Im super happy, Im shure it will be fixe soon.

* Does this bug happen in every project?
  
Its happen yesterday, first try, in an other project. Its happen today in a new project,
 I do it manualy now like before, but it will be amazing if it was working good. 

++++++++++++++++++++++++++++++++++

Nothing connect but does there is a topic somewhere, where we can ask for tips or question? To understand a bit more the tracker and also github.
Its firt time for me that im working with a sampler.
Im not familiar with the lexicon of the machine. Im shure I only use 10% of his capacity. 
Same for midi cc. First time. Dont know what is possible or not and dont understand anything in the midi implementation chart of my other gear....
 If some one have some time to answer to some technical question, it will be beautifull.
You can email me at groove.belliard@gmail.com to not overload this "issue" if you can take some time. I will be pleased to help you with one of my skill in return.

Best
Tom

talktogreg, 2021-01-21:
Hi

Just my 2c worth, but it could be that the TRS8 is taking time to register the Program Change message.

A quick Google shows that there may be a bug at Roland's end

    https://forums.rolandclan.com/viewtopic.php?f=56&t=56189

Cheers and good luck

Tombelliard, 2021-01-21:
Oh my god ! Really !!! 
Fuck its out since two year and half now. They sell I dont know how many TR, They will never fixe this shit ! They will never listen a fucking customer ! They do marketing brainfuck and profite !!! I do not deserve this !!!
I was so happy to have a machine who can control everithing and get my hand on something else than changing pattern !! with autosave, and the song mode with this super arrangement area.

I see you a developer, what do you think about this micro timming in negative value? Does it can fixe this stupid latency or it will create an other one

Tombelliard, 2021-01-21:
Oh fuck no this is no sens at all. The audio of the first step is suck by I dont know what

Tombelliard, 2021-01-21:
Sorry for my language

talktogreg, 2021-01-21:
I'm a developer, but not with/for Polyend.

Maybe you could add a suggestion post to have a default PC msg sent for each midi channel of Song load. This would mean the song would start OK. For latter PC changes, you could always use an empty step before the required note.

Tombelliard, 2021-01-21:
Yes I Guess this is whats i will do. Thé problème will be fix when changing drum machine one day, inchala.
Thanks for time and see you soon in a next tracker issue.
Best

Tombelliard, 2021-01-21:
The problem doesn't come from tracker

CliffRoberson, 2021-01-24:
I'm having this same problem. If possible, I would like to re-open this issue because it seems like the TR-8S could/should be able to get pattern changes from the Tracker without that audio stutter. The TR-8S can do pattern changes fine when doing them manually with the knob, so why can't we send the pattern change with that same sort of timing from the Tracker? This forum post (https://www.elektronauts.com/t/pattern-change-ot-from-tr8s/78611/20) seems like it might be describing a fix involving settings on an Octatrack where the program change is sent when the pattern changes as opposed to when a note is sent. I don't have an Octatrack but if someone does maybe they could test that out to see if it works?

CliffRoberson, 2021-01-25:
@Tombelliard I found a workaround that might help you. I was poking around on the internet and found this link with a workaround: https://www.elektronauts.com/t/roland-tr-8s-new-tr-8-with-user-samples-and-individual-outs/52578/1462. As it says, if you go to PTN SETTING menu on the TR-8S for the patterns you wish to use and change KIT SW to off, then the glitch you describe in this issue goes away. I'm not sure if there's a way to change kits mid pattern without a glitch but at least we can change patterns now.

Tombelliard, 2021-01-25:
Hey Cliff,
Thanks for your tip
I will check it as soon as possible. 
But lets reopen this issues is some one et some idea
Best !

closed 2021-02-07 by ambv

Brambo338, 2023-01-15:
I have registered my device in roland's website and i have started a repair ticket with this error.
The more people do this, the more chance we have for a potential solution.
It's a critical functionality issue and we need a fix for it.

sdesign75, 2023-01-28:
They have known about this bug for years and have done nothing about it. This is why I sold my TR-8S.

Brambo338, 2023-01-29:
I had a respond from Roland that the bug has been forwarded to the dev team.
Def not sure if it's going to to anything about it.
This sucks really bad Roland.

ozankasikci, 2023-10-01:
Has anybody found a solution? I've read on the internet that switching KIT SW to off would help but apparently, it doesn't. It's just unbelievable that Roland is aware of this issue and not fixing it...

ozankasikci, 2023-10-03:
For TR8S users who are having this issue, downgrading to firmware v2.0 solves this issue. Once again, a huge disappointment...