polyend/TrackerBetaTesting #722: [1.3.0] dB volume changes on tracker are double what is displayed

tinysynths, 2021-01-21:
I have thought for a while that when adjusting levels on the Tracker it feels quite sensitive and can change quickly and drastically.

I just tested this and in fact whatever dB value is shown on the Tracker display should in fact be twice that value.

Example: Track 1 volume on Master page is adjusted by -8 dB on Tracker. DAW shows that Track 1 is now exactly -16dB lower.

Example 2: Instrument 1 has level adjusted by -3.15 db on Tracker. DAW shows that Instrument 1 level is now -6.30 dB lower.

WugWugs, 2021-01-29:
I just got Tracker yeasterday, but I am confused with something that may be related behavior - when filling track with random volume values, for example from 5 - 55, the volume on higher values end up way too loud for what I expect (it sounds like notes around 50-60 are almost on full volume and clipping).

MortalJS, 2022-03-19:
No response. Closed

closed 2022-03-19 by MortalJS