polyend/TrackerBetaTesting #724: Play pause, resume from current position

phaelam, 2021-01-21:
...would be a nice addition.

talktogreg, 2021-01-21:
Doesn't shift-play resume?

phaelam, 2021-01-21:
Just shift? I was pressing shift+play. 
I swear I tried shift, I went rando trying. I’ll check tomorrow. My day has come to an end.

Just-Peter, 2021-01-22:
Shift-Play resumes from current position in Track view. Please close this issue.

phaelam, 2021-01-22:
you are correct sir.

closed 2021-01-22 by phaelam

hubparticipant, 2021-02-04:
A way to playback from the position the playhead was stationed previously would be a very welcome addition. 
It’s hard to explain. 
So if you put the playhead on step 17. A combination of let’s say LEFT+PLAY will start the sequence on step 17, regardless of the position of the playhead. Or an extra function button can be added: START FROM CURRENT POSITION / SET CURRENT POSITION AS START POINT FOR PLAYBACK. 
Add to this a loop playback selection. 

Currently Preview Selection works well enough though.