polyend/TrackerBetaTesting #727: [1.3.0] Sample Editor -> Broken EQ Fx?

anemlife, 2021-01-21:
Not sure about this but I was editing a few samples today and I'm almost sure that I hear no differences when cutting frequencies with the EQ! Also the fx processing was going way faster than it used to go for EQ in previous firmwares (as far as I remember). Could someone check if they also feel some issues with EQ fx?

thebradclay, 2021-01-21:
Agreed. I asked about the eq a while back and was told I was using it wrong. I've never gotten any results from it.

anemlife, 2021-01-21:
> Agreed. I asked about the eq a while back and was told I was using it wrong. I've never gotten any results from it.

Yeah definitely! I was expecting more drastic results, something similar to the 5 band EQ that we find on the master. (this one works perfectly in comparison)

I've been trying to cut/boost frequencies at maximum values with literally no results on Sample Editor.

thebradclay, 2021-01-22:
I tried cutting some bass earlier and didn't any change.

MortalJS, 2022-02-20:
Duplicate #422 
Closed

closed 2022-02-20 by MortalJS