polyend/TrackerBetaTesting #730: [v1.3.1b2] ghost midi note + step repeat = garbage midi data

somelamp, 2021-01-23:
This one is incredibly easy to reproduce:

1. on any track, enter any midi channel in the instrument column without a note
2. in either of the command columns on the same step, set a roll of any value
3. on the next step (or any in the same lane) set the note column to off
4. go to the performance page and set up a step repeat effect, and set the repeat to anything besides off
5. set the same track to be influenced by the performance fx
6. hit play and monitor your midi out

what you should see is just an insane amount of note on messages on the given midi channel, all playing the note 7C (E in the 8th octave) with a velocity value of 7F (the highest possible velocity value)

while there's not too many opportunities for this to happen in normal sequencing/programming, you can generate the same effect by accidently rolling a midi off message and simultaneously using step repeat. gave my blofeld (and me) a dang heartattack trying to produce some 48 note on messages in the 8th octave.

lemme know if y'all need any more info!

ambv, 2021-01-23:
This is a duplicate of #151 and is fixed. Await the next beta :-)

ambv, 2021-01-24:
Actually this is a little different.