polyend/TrackerBetaTesting #732: Grounding Issues

sharkytowers, 2021-01-23:
First- LOVE the Tracker.  Lemme say that straight off.  I realize this may not be the best place for this, but wanted to mention that I get really crazy grounding issues with the Tracker.  Probably because of the (really awesome) metal case?  When I use it in the studio I cant usually get the audio outputs to even work because of the issues (maybe the tracker automatically shuts off?).  I wonder if anyone else has had something similar?  Also, any tips to get around this issue would be awesome.

tilllt, 2021-01-23:
I solved the issue for my the ground issues of my notebook by using a TOSLink optical connection to the mixer, but i guess something like a behringer (i know, they suck) hd400 would probably do the trick.

sharkytowers, 2021-01-23:
great tip man. that unit should work. thanks so much!

[Sharky Towers - Chat @ Spike](https://spikenow.com/r/a/?ref=spike-organic-signature&_ts=ve15c)	[ve15c]

ancipital, 2021-01-25:
If you're talking about the buzzing noise it makes out of the analogue outs, that's not a ground loop (or at least it's not just ground loop issues)- I think it may be switcher noise.

I drove myself nuts trying to solve it, but it's present using a USB power brick, or powered from a decent-quality USB mains PSU plugged into an isolation transformer. If you notice, it's present in all the Polyend tutorial videos too. I mean, I suppose it's honest that they haven't doctored the audio to remove it in the videos, but you can be pretty sure that they wouldn't tolerate grounding issues in their setup.

My money is on a DC to DC switch mode PSU inside the tracker, which isn't sufficiently filtered/isolated from the rest of the machine.

sharkytowers, 2021-01-25:
> If you're talking about the buzzing noise it makes out of the analogue outs, that's not a ground loop (or at least it's not just ground loop issues)- I think it may be switcher noise.

No this is actually an issue where the unit wont power on, or have any signal out when using a 1/8" TRS to dual 1/4" outs into either DI boxes or basically anything in my studio.

ancipital, 2021-01-25:
> No this is actually an issue where the unit wont power on, or have any signal out when using a 1/8" TRS to dual 1/4" outs into either DI boxes or basically anything in my studio.

I have some faint memories of it failing to power up, but never made the mental connection. I think I must have unplugged it from the outs while trying to debug and forgot to worry about it later. 

I don't leave mine hooked into other gear due to the noise issues making recording the analogue outs unattractive. I also kept startling myself with how loud it gets on the outputs when you turn the tracker off and the muting relay kicks out, having to go mute those channels on the audio interface or mixer is irritating.

Now you've mentioned that, I'll probably notice it all the damn time :)

jaap3, 2021-01-28:
On a new project I also heard a constant noise. After muting the line-in, the Tracker was quiet again. I think new projects default to monitoring line-in, which seems to be noisy, even when nothing is plugged in.

drowsy-net, 2021-01-30:
> If you're talking about the buzzing noise it makes out of the analogue outs, that's not a ground loop (or at least it's not just ground loop issues)- I think it may be switcher noise.
> 
> I drove myself nuts trying to solve it, but it's present using a USB power brick, or powered from a decent-quality USB mains PSU plugged into an isolation transformer. If you notice, it's present in all the Polyend tutorial videos too. I mean, I suppose it's laudably honest that they haven't doctored the audio to remove it in the videos, but you can be pretty sure that they wouldn't tolerate grounding issues in their setup either.
> 
> My money is on a DC to DC switch mode PSU inside the tracker, which isn't sufficiently filtered/isolated from the rest of the machine.

here's what I found:

-  ground loop is isolated from the output stage

I'm guessing the output is on the opposite end of an I/O controller chip and the chip itself is properly grounded. 

- screen brightness affects the frequency of the hum

this points to RF interference at the USB port. I'd be curious to look at the signal path and figure out whether it's pure EMI from the gate driver of the display IC or if it's the DC itself that changes phase. DC power needs a resistor placed in series to an isolated power supply IC, or properly grounded to the same point as the I/O. The bus power and data stream should be separated here at the source. I can’t open mine up to measure more accurately. 

High kb/s transmission through the other ports affect the ground loop noise of the line-in (this is why the low bandwidth MIDI port doesn’t affect it). However I’m certain this entire input stage forms a circuit before it hits the controller. 

If you plug a 3.5mm cable to the line in and the mic in (lol) you can hear this looping charge pass through the cable, decrease in frequency until it oscillates at a lower pitch. When you unplug it, you can hear a “pop” from electric discharge.. the ground hum stops, then you hear it increase from zero all the way back up to the hum. So I think it's more than just interference but an actual mismatched voltage at the input stage.

ancipital, 2021-01-31:
> here's what I found:
>
>     * ground loop is isolated from the output stage
>
>
> I'm guessing the output is on the opposite end of an I/O controller chip and the chip itself is properly grounded.
>
>     * [...]

Hmm, "I can’t open mine up to measure more accurately" - it sounds like you're probably better placed to do so than I am :)

(Seriously, trying hard not to do something rash and get a decent modern scope recently, and don't need any encouragement.)

So I assume when you power the tracker down and the muting relay cuts in, that this accidental circuit is just a lot shorter and lower impedance? When the muting relay is.. muting, the output noise is noticably louder- e.g. when you power the Tracker down or during some stages of firmware updates. 

I am still struck by the irony of the muting relay increasing unwanted noise, even and sometimes especially on a powered-down unit. I don't own any other equipment that generates so much noise when switched off.