polyend/TrackerBetaTesting #733: [1.3.0b2] Unable to change desired parameters after hot switching instruments

b8ck8tt, 2021-01-23:
### Steps to reproduce

1. Add several samples to sample pool
2. Go to Instrument Parameters
3. Go to Effects
4. Change values for several parameters by holding corresponding Screen keys and turning the jog wheel
5. Observe everything working as intended
6. Press and hold the Instruments button
7. Choose another sample with jog wheel
8. Press and hold different Screen keys and try to change the parameter value
9. Observe that no matter what Screen key is being held, only the value for the last edited parameter for the previous instrument is being changed

### Attachments and links

Please find the reproduction steps here:
https://youtu.be/fBsxnmKBVe8

### Other important information

* How often does this bug happen?
Repro rate: 5/5

* Does this bug happen in previous firmware versions?
Didn't test

* Does this bug happen in every project?
Confirmed to happen in several different projects

The issue goes away after pressing navigation keys (left or right).

thebradclay, 2021-01-24:
Can confirm this happens to me as well in 1.3.0. and it happened in the previous fw release as well but beyond that I am not positive.  Yesterday I had this issue while adjusting in instrument Parameters. Press effects, make changes to overdrive, bit depth, reverb and/or delay. When I attempted to move to another parameter the red outline would move but when turning the jogwheel or used the arrow key to adjust the parameter the previously adjusted would move and not the selected one.

clickysteve, 2021-01-27:
Just coming here to report/confirm that this is still a bug in [1.3.1b3]. I was going to create a new report but saw this already existed.

closed 2021-03-11 by ambv