polyend/TrackerBetaTesting #737: Link a MIDI output channel to an instrument

brian3kb, 2021-01-26:
It would be useful for layering external instruments with the sampled instruments on the Tracker to be able to link one of the MIDI instruments to the sampled instruments.

E.g. I'd like to trigger tracks on a Digitakt at the same time as the instrument on the Tracker without taking up another track/step on the Tracker just to send the MIDI note out.

drowsy-net, 2021-01-30:
Excellent idea.

sunegg1024, 2021-02-08:
I was going to suggest this! Please - this would be so useful. If it can't be implemented as suggested, please add the ability to trigger MIDI in the effect column at least so we don't need to lose an entire track to midi sequencing. Thanks for the consideration!