polyend/TrackerBetaTesting #738: [1.3.1b3] Line In doesn't work in Sample Recorder

b8ck8tt, 2021-01-26:
### Steps to reproduce

1. Connect anything to Line In
2. Go Master, make sure external gear has signal coming in to PT
3. Go to Sample Recorder, choose Line In as your source
4. Observe PT unable to detect the same signal

### Attachments and links

https://youtu.be/5JuCZPz2LR0

### Other important information

* How often does this bug happen? (every time, often, "it only happened once")
Repro rate: 5/5

* Does this bug happen in previous firmware versions?
Confirmed to be introduced in 1.3.1b3. I rolled back to 1.3.0 and 1.3.1b2, and everything did work there.

* Does this bug happen in every project?
Bug was reproduced in several projects

Please notice if you plug the same cable to Mic In, it's being detected in Sample Recorder, only Line In doesn't work.

b8ck8tt, 2021-01-27:
As reported by Discord users arcologies and somelamp, Line In sampling works for them in 1.3.1b3.

b8ck8tt, 2021-01-27:
Cannot reproduce the issue anymore without any specific actions towards it's fix.
The last time the issue could be reproduced I went through checking 1.3.0 (no issue), 1.3.1b2 (no issue), 1.3.1b3 (issue reproduced) - all on the same project (don't remember if I created it on b2 or b3). Then while staying on 1.3.1b3 I loaded another older project (most probably created on 1.3.0 or 1.3.1b2), also reproduced the issue. Powered off the unit. When I turned it back on, the issue was gone.

birdydy, 2021-01-27:
I have exactly the same issue, happens every time (I happen to also use a microfreak). I was wondering if it could have something to do with the fact that the input is stereo, and we are inputing a mono signal. 
If I connect my microfreak with a mono (TS) cable to the Line In of the tracker, I hear the sound on the left only (meh), wether monitoring or simply playing my track  - but at least the recording is working. 
If I connect my microfreak with a stereo (TRS) cable to the Line In (is that your case @b8ck8tt  ?), I hear it in stereo monitoring / playing (cool), but the recording is not working, as you describe. That must have someting to do with the signal flow in there. 

Fixing this (or explaining how to avoid it ;-)) would be cool because it is not very convenient (though not the end of the world) to switch between line in and mic in when switching from playing to recording.

tfspsound, 2021-01-28:
[1.3.0] I noticed recording earlier today that the PT is ignoring one of the input channels.
It would be nice to have the recording option to record Left or Right and the option to have a mixed down mono version of what we can hear in stereo. 
Currently we can’t sample stereo samples meaning anything panned hard to whichever side is ignored will not be recorded.

Just-Peter, 2021-01-28:
> [1.3.0] I noticed recording earlier today that the PT is ignoring one of the input channels.
> It would be nice to have the recording option to record Left or Right and the option to have a mixed down [...]

Just like the Synthstrom Deluxe, choices for left, right or stereo.

bichan17, 2021-01-28:
> [1.3.0] I noticed recording earlier today that the PT is ignoring one of the input channels.
> It would be nice to have the recording option to record Left or Right and the option to have a mixed down [...]

I have a ticket open for this here: #548

b8ck8tt, 2021-01-30:
To be clear, I'm connecting Microfreak (mono output) with a stereo cable to PT's Line In.
I'm still experiencing an issue on 1.3.1b3, and it doesn't go away, no matter what I try to do.

tfspsound, 2021-01-30:
Even if you use a mono cable the issue persists? If you haven’t got a mono cable try swapping the L and R connections into the adapter if you are using that?

b8ck8tt, 2021-01-30:
It did work with a mono cable. Also with a stereo one I unplugged it almost all the way out, but not entirely, so it was barely making contact to MF's socket, and it started working as well.

Overall, the current understanding of the issue is that if a synth's mono out is being connected with a stereo cable to PT's Line In, PT might not recognize the incoming signal.

tfspsound, 2021-01-30:
I’m guessing they aim to do something about this or they wouldn’t have put a stereo input on the PT in the first place.