polyend/TrackerBetaTesting #739: Enhancement: "Faster" single-track Live Record Mode (truncate/interrupt previous notes)

ConnorCarbon, 2021-01-27:
## Normally, Tracker records overlapping MIDI input to "spill tracks" to the right if a prior note on a given track has not sent an OFF message (and/or the instrument/sample has not ceased playback. Not sure 🤔), and if there is no other track available (like track 8 or when user constrained armed tracks for recording), it will just not "allow" more notes to truncate or interrupt prior notes.

This is logical and good in "slower" live recording cases with the on-board pads as well as with external MIDI controllers. However, in the case of, say, arpeggios or fast percussion entries (or really anything rapidly triggered during a live record take), a user will have their "part" across 2 or 3 (or more!) spillover tracks and will then have to manually clean up that data to consolidate onto less tracks (ideally, one). This usually happens because, even when playing arpeggiated chords cleanly and quickly, a user's fingers cannot leave a prior pad quick enough before engaging the next, so the OFF note left behind usually prevents the next note to trigger onto the same track (or at all, depending). In this case, the user values the ability to preserve the "rapid performance" note / velocity / microtiming data more than the preservation of accurately-programmed OFF messages in the pattern.

### Suggesting: Live Record "Truncate" or "Interrupt" or "Rapid Input" Mode.

This mode would instruct Tracker to ignore waiting for OFF messages and/or the cessation of a sample before allowing another triggering of said sample (or any sample) on any track currently armed for recording, but especially on the track(s) that the user intends to keep all of a live record take's data in, without "spillover" or unnecessarily exempted notes from a take's performance due to a lack of OFF message or sample "overlap". The idea is to allow Tracker to truncate prior triggerings of a sample and immediately begin anew, regardless of how quickly the user re-triggers.

(Im a new-ish user so if such a feature already exists, I apologize and do let me know.)

Otherwise, I wanted to add that, often, the tail of one-shots and samples that seem to cause a "blocking" of arpeggiated notes on the same track or in rapid succession are usually near-silent anyway, and if this Mode did run the risk of "choppy" re-triggering of a same sample during rapid love performance takes, there could be an automatic 5 millisecond fade out of the truncated sample, or similar to mitigate that choppy interruption due to the ignored OFF release of the performer's finger or etc.

This would mean that anyone who can cleanly play 16th notes or other rapid MIDI triggering within a pattern could fit all their intended performed notes into their intended armed recording tracks on the Tracker regardless of how long the sample was or if they lifted their finger / drum stick / hand / etc. in time during arpeggiation, stick drumming, MIDI Guitar In, finger drumming, experimental MIDI control, etc!

birdydy, 2021-01-27:
Would love that as well (or something leading to the same result). I find live recording  the area where the Tracker can be most frustrating (one could say it's not core to the Tracker spirit, but still). I am probably not very good, but it can take me a ridiculous amount of time to properly record even some simple figures.

ConnorCarbon, 2021-01-28:
> (one could say it's not core to the Tracker spirit, but still). 
Yes! And it's totally okay that the Tracker has this going in in its first year of launch, because Polyend, I believe, set out to develop a contemporary retro device, yet accidentally made one of the top contenders in today's "groovebox market"

Here's a link of my little spreadsheet I made to compare it to similar products to show how well it does compared to them! (it's subjective to my preferences and situation, but still!):
https://docs.google.com/spreadsheets/d/1MieOTU4SrKeH45-9RTYPBlgbNNUb2zKCJMkVrix1cP8/edit?usp=drivesdk

gmli, 2021-01-31:
It indeed doesn't exist in the Tracker. And for me it's THE major limitation of it for playing in « mono ». But it has nothing to do with fast playing. Currently it's impossible, even at ultra slow tempo, to play legato in mono track live.

johnpink, 2023-01-04:
> Here's a link of my little spreadsheet I made to compare it to similar products to show how well it does compared to them! (it's subjective to my preferences and situation, but still!): https://docs.google.com/spreadsheets/d/1MieOTU4SrKeH45-9RTYPBlgbNNUb2zKCJMkVrix1cP8/edit?usp=drivesdk

That is by far THE most beautiful spreadsheet I have seen in my entire life. Props to you. Good day :)
Also, +1 for the feature obviously!