polyend/TrackerBetaTesting #740: Polyend as a master sequencer; Global 'send program changes' setting

jimmac, 2021-01-27:
I love how the Tracker can sequence other gear via MIDI. I also learned that you can send MIDI program changes for other gear to change patterns. However manual pattern change sequencing is laborious. I'd love to propose a global 'Send Program Changes' setting in the MIDI section that would make the Tracker send program changes automatically based on the pattern order in the song sequence.

# Use case
While the current 'do everything on the Tracker' is a viable option, there are constraints of a 8 track pattern used for both native instruments and MIDI intsruments. Offloading pattern sequencing to other gear will also allow to use specific workflows such as conditional trigs (other than chance that polyend also supports). 

# Suggested workflow
What I find particularly appealing is a workflow where I lay down the bits and pieces on the digitakt and digitone, allowing to jam the tune live too. Once the track solidifies, I'd lay down the sequence in the Tracker with empty patterns to be filled with additional percussive sounds, vocals and effects. And for the final DAW-less workflow step, let the Tracker play back the sequence, allowing me to adjust the filters and synth parameters live, without 'baking in' any automation.

birdydy, 2021-01-27:
That would be great, generally. But as far as the idea is to use it specifically as the Tracker a song master for a digitakt/digitone (which I'd love, having in perspective exactly what you describe), I'am afraid you'd run into the annoying problem of those boxes not reacting immediately to PC messages :(

jimmac, 2021-01-27:
> reacting immediately to PC messages :(

That is indeed the case. You have to give the elektron boxes some time to react. I will do some testing as to how short that can be. I've settled on putting it on the equivalent of a 13th trig on the pattern, but I'm sure it can be shorter.

birdydy, 2021-01-27:
But then how would what you imagine (a kind of automated PC sender in song mode) handle this necessity of sending the PC messages in advance ?

jimmac, 2021-01-27:
Perhaps there is additional way than the CC messages. On Elektron gear with their global 'PROG CH RECEIVE/SEND' the change is immediate. 

https://user-images.githubusercontent.com/14821/106032103-575b5700-60d0-11eb-8668-70fb457cbe04.mp4

Just-Peter, 2021-01-28:
I really like this feature request! Hope Polyend can add this.

alexanderflink, 2021-02-07:
I would love to see this as well. Currently I am creating an extra duplicate pattern for each change just to put a PC message at the end of the pattern (the Elektron boxes seems to need roughly two steps to react to the PC message).

andyl-astardigital, 2022-05-01:
Would love this too!