polyend/TrackerBetaTesting #741: [1.30] Tempo Fx per step stop working in external clock mode

sunfallz, 2021-01-27:
Tempo Fx per step in patterns cease functioning when in external clock mode. 

Reproduce: 
- I have 2 trackers  (of course any gear sending CLK)
- 1 is the master clock, the 2nd tracker set to EXT CLK, tempo fx stop working in this mode on the receiving tracker

* How often does this bug happen? Every time
* Does this bug happen in previous firmware versions? Only tested in 1.30
* Does this bug happen in every project? yes

MortalJS, 2022-02-20:
Hi @sunfallz 
Can you check if the problem occurs in firmware 1.5?

sunfallz, 2022-02-20:
Hi @MortalJS :)
Confirmed in 1.5:
Sending clock from A4 mkII -> Tracker in external clock/transport mode, Per Step TempoFX are ignored, still have no effect.

MortalJS, 2022-02-22:
Still in 1.5. Bug confirmed.
@ziej8asek