polyend/TrackerBetaTesting #742: 1.3.1b3 Held down cursor speed is too fast - Step cursor movement

wraggal, 2021-01-28:
In the latest firmware the held cursor speed on the pattern view has increased a lot, which is far less usable. holding for more than a second makes the cursor fly off in an uncontrollable way, leading to lots of single button presses instead of held down cursor movement. If this was a previously requested change it's dialled in too much, or should be optional/adjustable.

A good optional feature might be to allow the cursor to follow the 'step' value at all times.

tfspsound, 2021-01-28:
#438 is another request for this I’m sure there are a few more too, it’s not easy to search for cause ‘step’ is such a commonly used word on here.

UserOverflow, 2021-01-28:
I just would to request for keyboard configuration. One setting for "speed" and second for "delay", similar to traditional keyboard. I was using trackers and it was always fast and without delay and it was great ;) But with unwanted delay (or delay value) I still have problem to feel the same speed with tracker :)

ConnorCarbon, 2021-01-28:
Yes! I've noticed that "press and hold" of a key is too slow to accelerate when scrolling up a list at a "medium pace" but the jog wheel used for scrolling tends to be extremely too fast and go too far when turning it quickly.

Also, I wish in the sample editor, when setting in/out points (including on the "trim a radio recording" screen which is basically the same thing), that the jog wheel still had its multi-speed functionality. It can take forever to move in/out points and I think also splice markers with the jog wheel, currently, even when zoomed out all the way. If that counts as a different issue let me know and I'll create one

closed 2021-02-27 by ambv