polyend/TrackerBetaTesting #744: "no interpolation" Settings

gurkenkimchi, 2021-01-29:
I’m wondering if the tracker includes options to disable sample filtering when transposing samples, as I prefer using “no interpolation” settings in samplers to retain the aliasing you get from playing back .wavs at a different rate. Are there options for this, or can anyone tell me about the filtering method used?
i’m just looking for a way to retain the original crunchy sound of the samples, as opposed to the filtering you get when transposing them on the tracker

a basic interpolation settings per sample. even a basic on & off switch would work.

tfspsound, 2021-01-29:
I think there is an aliasing option in the Config options on [1.3.0]

drowsy-net, 2021-01-30:
Agreed!!

The "bitcrusher per instrument" effect is not enough for me when my goal is to take 8-bit samples and transpose them down by at least an octave (even 3!). The Tracker already converts samples into 16-bit WAVs upon import so I'm not very optimistic that the Tracker is designed to handle crunchy samples. The interpolation engine is so different from my Amiga :(

When the Amiga interpolates, there is still clarity on the high end and much less ringing. It doesn't just artificially raise the noise floor like this algorithm. I imported drum breaks sampled from the A1200 and when I transferred them to the Tracker they sounded significantly worse when pitched down. Muddy, jagged-edged and thin. 

The bitcrushing effects only seem to toss out extra information without any real change in how the Tracker handles the sample file. I expected it, but it's such a niche issue to request as a fix. I hope they just throw us all the options to fine tune this ourselves. 

Everything from sampling -> filtering -> interpolating -> mixing -> final rendering is affected by the bit depth of the project. Even the reverb algorithm would change in texture if it was doing math in different bit depths / sample rates. Some of us would prefer more control over that texture. A man can dream!

theorize999, 2021-02-13:
I'm pretty sure there is an option already in the settings for whether or not to use interpolation, but a real improvement would be to have that option on a per instrument level. for single cycle or other short looped samples no interpolation might be wanted but then say a sax sample in the same project might want interpolation applied.

trutypesounds, 2021-07-21:
+1 for on/off interpolation per sample