polyend/TrackerBetaTesting #746: [1.3.0b3] Tempo FX per track not working for external midi sequencing!?

Just-Peter, 2021-01-29:
Track 1 : Note C4 M01 T10 
Track 2 : Note C4 M02 T20

I expect that Track 1 keeps playing at T10 and Track 2 at T20, so Track 2 is the double tempo of Track 1.

Or is this not possible? Or do I do something wrong?

* How often does this bug happen? Every time

* Does this bug happen in previous firmware versions? I don't know

* Does this bug happen in every project? Yes

Just-Peter, 2021-01-30:
Seems to be this is not possible (yet?).

sunfallz, 2021-01-30:
It is only possible to have 1 track with tempo fx, as it effects the whole patterns bpm. 
Same reason that pattern length fx applies to the whole pattern. (i.e, we currently can't have individual track lengths)
hope this helps.

Just-Peter, 2021-01-30:
> It is only possible to have 1 track with tempo fx, as it effects the whole patterns bpm.
> Same reason that pattern length fx applies to the whole pattern. (i.e, we currently can't have individual track lengths)
> hope this helps.

Thanks!

ambv, 2021-02-26:
Yeah, T is a global FX.

closed 2021-02-26 by ambv