polyend/TrackerBetaTesting #748: [1.3] Sample Effect, Equalizer

positionhigh, 2021-01-30:
The Sample Equalizer does not seem to work, either in the preview, neither in the final rendering.
Doing massive +-12 db changes, seem to have no effect on the sample.

### Steps to reproduce
load sample, make drastic eq changes - changes seem not to apply at all.

thebradclay, 2021-01-30:
Similar to issue #727 . I asked about the eq issues a couple months ago as well on the fb group (I think that's where I posted).

topoexplo, 2021-02-21:
I’ve noticed this as well. It either isn’t working or it’s so subtle that it’s imperceptible.

thebradclay, 2021-02-21:
@topoexplo 

I'm asking about it in the discord to see if anyone can offer some assistance.....here is what I asked:

Hi, can someone explain the sample editor equalizer?  It either doesn't work right or I'm an idiot. I understand how an eq works but this is strange.

I've had issues with the eq since day one and just don't use it and get by with filters but that is frustrating and doesn't help most of the time.

So, I am using the sample from /arc noise/bass(1-100)/bass_808_bash_F.wav 

Then in sample editor select eq. completely cut the low to -12.000 dB. apply, look at and listen to sample, sample, hit undo notice there was some visible change in the wav form but not really perceptible.

Now with the original version of the sample select eq,  boost the low to 12.000dB, apply look at and (maybe don't) listen to sample, hit undo and notice there was a visible change in the form.
So what do I need to do to properly cut out low end from samples using this eq

trutypesounds, 2021-06-20:
same thing here, 1.3.1 eq isnt working in a proper way.

MortalJS, 2022-02-06:
#422 duplicate

closed 2022-02-06 by MortalJS