polyend/TrackerBetaTesting #749: [1.3] more Channels or Midi Channels

positionhigh, 2021-01-30:
## Hi, we're excited to hear your idea!

I understand that the concept or a tracker is limited to a number of tracks that requires cpu power. We already have more than enough RAM for samples, also there are more than enough slots for samples (and yes, this should be very short samples). Can you include, optional, more Tracks, if this is not possible for Tracker/Audio Tracks because it is bound by CPU power, maybe it is possible to to add more Midi Tracks. 4-8 Midi Tracks would make a big difference.

Just-Peter, 2021-01-30:
Maybe I do not understand your request, but you already have 8 midi tracks?

positionhigh, 2021-01-30:
Yes that is true and i also understand i can add midi chords using only one of the tracks.
Coming from a tracker background but working the last years with DAWs only, the limitation to 8 tracks feels more limiting than i thought it would be. 
I see that this kind of limitation is a part of the product strategy to not get distracted or having too many options as in a DAW.
Nevertheless, for me it would be a big advantage to have a option to use more than 8 tracks.
I asume the GUI implementation would not be too difficult. In the tracker view, it screen is already scrolling and displaying only 4 of the 8 tracks. It should be possible to extend that scrolling. For all other relevant pages, like mixer, it would be fine to swap between pages 1/2, 2/2 either by holding shift or just cycling through 2 pages, like it is done for other functions already.

My thought is that the tracker could reach a much wider audience this way from people coming from DAWs and not only attracting users with nostalgic tracker background. For those who dislike this idea, this all could be made as a option in the setup like for example "tracker mode" and "pro-tracker mode".
The main question is if the hardware could deal with the extra load but i suspect this could be possible.

theorize999, 2021-02-13:
I also feel that 8 tracks is a little TOO limiting, and honestly only the oldest of trackers are limited to 8 or fewer tracks. I suspect cpu is a limiting factor here. May I offer a possible solution? Would it be possible to add more tracks that don't offer as much features? such as maybe tracks 9-16 could be drum tracks that don't offer any granular, wavetable or filtering/realtime effects... only basic sample playing and send effects. The drum tracks should have slicer capabilities though.

tilllt, 2021-04-03:
I think this is a duplicate of 
https://github.com/polyend/TrackerIssues/issues/640

theorize999, 2021-04-04:
I'm against putting MIDI only tracks in, I feel this unit as a tracker has been compromised enough. Apparently we will have a tracker that goes sideways but still no arp command. Still no slide commands because nobody ever wanted to sweep a filter ever. I hate to be negative but I'm not really all that happy as to the focus in development lately. I don't make chiptunes so it's absence doesn't hit me that hard... but I still think it's a command every single tracker should have. Maybe have this thing be a solid tracker then worry about the gimmicks. If polyend doesn't be careful they're going to have a machine that does a bunch of things, but doesn't do any of them well.

positionhigh, 2021-04-04:
My primary wish was to be able to use more synth/sample tracks than 8. Having more midi tracks was more of a side-question since this should not be very ressource intensive to the cpu. I am not sure if i read correctly but to my current understanding we are dealing with a cpu similar or identical to a teensy 3.6. I have no idea if this is also in control of the tft. if so, the machine might be near its limit. if this is handled otherwise, it should be possible to give the option to use more sample tracks.
to the comment of theorize999: it is sad that tracker isn't able to reproduce even simple 4 channel mod files faithfully / with most features. when i had bought it, i had the impression this should work without even thinking about it and even extend around this concept. This this is not the case, isn't much of a downside for me personaly. My biggest wish is, as was the original reason for this topic, to have more than 8 sample channels.

bambrose, 2021-04-04:
I do agree with the sentiment that the introduction of extra features which detract from what I think is the core of this machine (on board instruments and sample manipulation) is a shame. Currently the workflow and interface is focused fun and fast but has some important omissions (saving / loading instruments for example)
I'm not sure that the introduction of midi only tracks is such a detraction from this core use.. it does rather depend on whether it would represent reducing cpu which could be deployed elsewhere I suppose (like a "volume decay" effect for example)

theorize999, 2021-04-05:
I apologize, that was just me venting frustrations, I actually would benefit greatly from extra midi channels. It already pairs well with my blofeld as is, a set of midi channels to send to it would really elevate the utility of the box for me. I just think polyend needs to really improve the core tracker to really live up to what it proposes to be, no matter how cool it would be for extra midi channels to happen, it needs to at least be able to do the basics.

tilllt, 2021-04-05:
For me, a sample pattern page with 8 Tracks and a midi pattern page with 8 tracks would be a clean simple solution
1. to extend the amount of overall tracks
2. to have a very logical and clean seperation of midi and internal sound

mstempell, 2021-08-14:
If channels are added, it would be a shame if these were be limited to MIDI.

tilllt, 2021-08-14:
The tracker is at its IO limits as is. Since a midi channel requires much (much) less performance than an audio channel, extra midi channels might (or might not) be feasible. Extra audio channels are definitely, not that has been stated clearly.

mstempell, 2021-08-14:
Ah OK, thanks for notifying me about that.