polyend/TrackerBetaTesting #758: Sample Generators

mwmwmw, 2021-02-01:
Back in the '90s, when I was a broke teenager, I used to have a ton of little 'generator' programs instead of actual hardware. 

If you're as old as me, you'll surely remember Smorphi, Stomper, FX2, and FM Dreams. See ['non-realtime synths' ](http://www.maz-sound.com/+soundtrackers/synths.html)

It would be cool / fun to include something like that in the Sample editor/recorder.

I propose 3 "offline" renderers. These would render a sample and not be real-time. 

1)  Single-cycle additive waveform generator. Add/Subtract sines at different frequencies to produce a single cycle. 

2) An offline synth with 3 detunable oscillators, with wave selection (sin, tri, saw, square, noise) and amp/pitch envelopes. With that, you can generate a ton of sounds. Including percussion. 

3) Etch A Sketch Mode: Trackers usually let you 'draw' into a table. But, that's not possible. Maybe an 'etch a sketch' mode where you run a play head (x-axis, up/down arrows control the speed) then use the jog wheel while holding down 'rec' to draw the wave in (y-axis). Add a little cursor to show where the 'write' head is. Punch in to smooth out errors.

anemlife, 2021-02-05:
This would be a total game changer! I 100% agree, something to generate sounds other than the radio would be great!

hubparticipant, 2021-02-07:
The idea of the Etch a Sketch got me all itchy! How cool would that feature be? It’s brilliant !
It fits the idea-présence of the Polyend Tracker 100%. Brilliant

And a simple barebones offline waveform generator: also yes 👍🏼

hubparticipant, 2021-02-07:
I have another one:

- Screen keys represent 8 different waveforms. Pressing one or several will start playback of the waveform(s). 
- REC starts recording
- Jog wheel adds harmonics
- Holding down one of the waveforms (or multiple) + UP/DOWN will change semitone by 1. LEFT/RIGHT detunes the waveform(s)

theorize999, 2021-02-13:
What we really need is a single cycle generator that is like the one in Klystrack, the additive would be welcome as well but the generator in Klystrack is much faster to work with, and also a full PADsynth algorithm with most of the options found in ZynAddSubFX (the creator of PADsynth)... most of the other PADsynth implementations I've seen are way too simple. Using these two tools along with wavetables really increase the synthesis capabilities of the tracker since we can resample (which should be more streamlined with an option to render directly to an instrument instead of to the SD then loading it). I think a good idea for an offline synth would be something akin to Orangator, it pretty much had all the bases covered, versatility wise. It was and actually still is a great synth, especially paired with polyend tracker since it doesn't use multisampling and outputs mono samples. I've actually started using it again, to have something like it on the tracker would be beyond awesome.

Anexium, 2021-07-24:
I like your idea of the single-cycle additive synth.  To be honest, just a quick way to get a basic waveform (sine, square, etc) would be really handy - the additive aspect would be a bonus.

closed 2022-05-19 by Jacknametrouble