polyend/TrackerBetaTesting #762: We can control global sample playback volume in the sample loader.

modmodmodmod, 2021-02-02:
I've made similar requests in the past. The content was that the preview volume of sample was too loud compared to the playback volume of song, so please lower the preview volume of sample. And I received some advice. The content was basically to raise the playback sound of the song. 

However, I think the volume of the song should be set so that each track doesn't clip with the master leaving a little headroom. And if I do so, the sample set to today's general volume by splice etc. will be too loud in the sample loader. 

So I think it would be very stress-free if I could raise or lower the sample preview volume globally at sample loader, so I'd like to request that feature.

somelamp, 2021-02-03:
honestly, I can't even consider this to be an "enhancement" at this point, this is borderline a bug.

the fact that the only thing controlling the sample preview volume is the master volume control is wild. it's one of the first things I noticed when I plugged in the tracker for the first time, and it still plagues me every time I sit down with to make music.

as modmod has saying, it's really easy to have levels entirely too hot and leave yourself little to no headroom to work with on this device, so I typically lower all of my levels in both the instrument parameters and master sections. however, anytime trying to preview a sample is nearly aggressive at that point.

if "project memory" on the sample loader page is shrunk down to a single button space, there would be plenty of room to put a controllable preview volume level right next to it. or, alternatively, it could just be a global/project setting in the config.

 in tandem, it'd also be nice to see the default instrument volume lower than 0db. I feel like you'll find similar in most software and hardware based workstations and instruments. I probably wouldn't be so quick to pull down my levels on the master page were that the case.

astrochili, 2023-03-22:
I think there is nothing wrong with previewing samples at 0.00 dB if you use another way to control the sound, such as an audio card.

Perhaps the problem is the lack of a volume control specifically for direct headphone listening? I think such a parameter in the global settings (not project settings) that controls the output volume level would be a great solution that does not violate the concept of zero level and allows you to work with music in headphones.

That is, I propose on the contrary, not to reduce the volume of the preview, but to allow you to adjust the overall output level from the device (not the project) in deep device settings.

Translated with www.DeepL.com/Translator (free version)

astrochili, 2023-03-22:
Already is here: https://github.com/polyend/TrackerBetaTesting/issues/1120

astrochili, 2023-03-23:
Sorry guys for the confusion, I've figured out that the Master level affects the preview volume as well, so overall everything seems to be ok.