polyend/TrackerBetaTesting #763: Multiple slots selected in song mode can be played in a loop.

modmodmodmod, 2021-02-02:
Currently, I think that only one slot can be played in a loop at song mode may be. However, I often would like to loop play two or more patterns in succession, so I request this feature.

ketumatek, 2021-02-02:
Plus One..
It will be nice because at this moment we can loop only one slot ...
As we can select two or more we should loop them ...because when you are composing something you need to loop 2 or more slot to see what is going on...

topherdavis, 2021-02-12:
I'm trying to decide if I should keep my NerdSeq or Polyend Tracker and the only feature that has me completely sold on the Nerdseq is "islands". This is basically what they are asking for here.

On the song mode you can group multiple patterns together that will play one after another and then loop. You can use this to create longer patterns.

"From here islands or chains of patterns are combined to build up loops or songs."

[image]

To create the islands you leave gaps on the sequencer screen or on the tracker song mode. The nerd seq also has the ability from this screen to select an entire row to play or you can just select one of the islands. This is really valuable for performing. You can create combinations of the patterns you've built up and remix the track live much more easily than I can on the tracker with the method of switching patterns live.