polyend/TrackerBetaTesting #765: Sampling / previewing left or mono input channel

hubparticipant, 2021-02-03:
Summing left and right channel into a mono file is not always a good thing and may even, in some cases, lead to phasing issues. 

Currently when connecting a mono mini jack only the left input is heared, in your left ear only. 
This is not very convenient. 

I would like to have the option to preview (and record) either the left or right input channel as a mono signal. Where it is heard in the middle of the stereo field.

BlakRainbow, 2021-02-06:
I would also like to see an option to monitor incoming signals in mono, most hardware I'm sampling is mono and having the audio playback on only one side is annoying.

hubparticipant, 2021-02-06:
It is very annoying indeed

AlexNorinh, 2022-04-28:
Another take on this - to be able to route 2 mono synths or other external devices centre panned for monitoring/line in FX