polyend/TrackerBetaTesting #768: Request: Set current playhead position as start point for playback

hubparticipant, 2021-02-04:
Would be nice for editing long patterns. 

Add a function button with the following action:
SET CURRENT POSITION AS START POINT FOR PLAYBACK. Which is a toggle button.

Just-Peter, 2021-02-04:
Doesn't shift>play do that?

hubparticipant, 2021-02-04:
Hello Peter
Shift + play will start from current playhead position. It will start playing the pattern where it was stopped. 

Let’s say you’re editing between step 96 and step 108. To hear your edits, you have to put the playhead back to 96 in order to hear your edits. 

Feature request solves this with a extra function button: PRESSING PLAY WILL ALWAYS START PLAYING BACK THE SEQUENCE ON STEP X if toggled on.

taytotim, 2021-04-06:
Adding to this, apologies if this is already possible, I cannot find out how. Essentially idea is to to be able to loop part of a pattern. Adding looping to preview selection. If you’re editing a fiddly section of a pattern it would be great to be able to keep editing and have it loop while you do it. Currently have to go back to the start step and replay, or wait for that section? Unless I have missed the functionality.

thanks

pinkcell-niico, 2021-04-06:
Oh yeah rock 'n roll, that misses me so much to be able to loop the editing part !!

yoshihiroxx, 2024-02-14:
This feature is so nice idea. That's my want. Please give it us.
I googled "polyend tracker playback" to finding how to do it.