polyend/TrackerBetaTesting #769: [v1.3.1b3] Scrolling speed (unsure if bug or intentional)

thebradclay, 2021-02-04:
Last night I noticed that when scrolling with the arrow keys the speed is ramped up much higher than before. Is this intentional? It makes stopping where is needed difficult. I thought I might be imagining this so I rolled back to 1.3.0 and it is indeed different.  I would like there to be some way to scroll both slow and fast but if I have to choose I would like the previous speed.

To recreate this, be on v1.3.1b3 and scroll through a pattern by holding down  the arrow up and down keys. Notice the speed ramp up to hyper speed.

Roll back to v1.3.0 and scroll through a pattern by holding down arrow up and down keys. Notice the speed stays steady.

* How often does this bug happen? every time in v1.3.1b3

* Does this bug happen in previous firmware versions? No

* Does this bug happen in every project? Yes

bichan17, 2021-02-04:
As per discussion on the PT discord, I believe this is intentional. I do agree that it goes too fast at the moment.

b8ck8tt, 2021-02-09:
Can it be a setting in the menu, please? Maybe several options to choose from, or even better - a scale from 0 to 10, that would translate to a scroll speed modifier, whatever it is under the hood?

wraggal, 2021-02-10:
I noted the same here [https://github.com/polyend/TrackerIssues/issues/742]

ambv, 2021-02-26:
This is being solved as part of #742.

closed 2021-02-26 by ambv