polyend/TrackerBetaTesting #771: [1.3.1B3] Midi instrument lane does not delete when deleting note

rpc9943, 2021-02-05:
## Hi, sorry you had trouble!

### Steps to reproduce

1.enter in some notes under a midi channel on instrument
2.delete some notes, notice that MXX stays and does not delete when deleting note. Unlike when you delete a sample, it will erase the instrument number.
### Attachments and links

Often the best way to show the problem is to record a short video,
attach the Tracker project you were working on, and provide links.

Pro-tip: You can drag and drop images and `.zip` files directly to the edit
box when writing your bug reports. Videos are too big as attachments, you
will have to use an external service for this and provide us links. Please
don't use services that delete files after short periods of time.

### Other important information

* How often does this bug happen? (every time, often, "it only happened once")
every time

* Does this bug happen in previous firmware versions?
unsure

* Does this bug happen in every project?
yes

ambv, 2021-02-07:
This is deliberate, so you can send MIDI CC without triggering notes.

closed 2021-02-07 by ambv

rpc9943, 2021-02-07:
How should we erase midi notes? Thanks!

birdydy, 2021-02-08:
Simply press "delete" when on the "intrument" lane ;-)