polyend/TrackerBetaTesting #773: Suggestion --> Perform mode: option to record the output as a "stem"

anemlife, 2021-02-05:
It would be crazy if we could record the output of the master when playing in Performance mode with let's say a pre-defined limit in steps for every recording (128/64/32 step recording, etc...). Not sure about the way this could work but it would be super useful to manipulate the samples in a deeper way.

Example: 

- I set the Perform mode as I want (choosing each pattern for each track, muting the ones I don't need etc...)
- I set the Perform parameters as I want them. (step repeat, reverse, etc...).
- I press on "record performance" (no idea how this could be implemented)
- The tracker asks for the length of the recording in steps. I chose the one I want.
- I press play, perform the pattern as I want, it gets recorded and we're done!
- The file could then be renamed as we want and finally get saved to the export folder in a "Perform" subfolder!

Thanks :)

synthexpert, 2021-02-05:
This would be a huge improvement! Please you guys at Polyend add this :)

tfspsound, 2021-02-06:
I think I read a similar idea to this towards the end of last year it’s a neat idea. I’d like to see a second performance screen with another grid where you could store ‘presets’ of performance mode fx combinations so you could change all fx simultaneously to your own predefined configurations.

closed 2021-12-02 by Jacknametrouble