polyend/TrackerBetaTesting #774: External midi controller - request for on+off for Mute/Solo/Rec

danapeks, 2021-02-05:
Hey there 

Please could we have the option to map on/off midi cc to mute/solo/rec functions on controllers like the Korg Nanokontrol or Novation Launch control XL? Be super helpful so we can use when doing live performances. 

Alongside this be great to be able to set maximum CC values for external controllers - am finding volume faders are tricky to use on my nanokontrol as pushing them fully up pushes the vol super high, be cool to be able to limit it without setting the vol of a channel really low. However, if could just use the mute function on the nanokontrol this would not be as much of an issue

Thanks

phaelam, 2021-03-22:
I second this. MUTE CC would be smurvey

static-noise-PT, 2021-05-01:
Yes mute and solo at least please

closed 2021-12-02 by Jacknametrouble