polyend/TrackerBetaTesting #775: midi note length

BlakRainbow, 2021-02-06:
I would love to have the ability to choose the note length when sequencing external hardware synths, having to program "off" msgs in-between notes in general is time consuming and makes the whole process feel less musical. 
It would also save time from having to add "off" messages in-between notes when using FILL mode.

hubparticipant, 2021-02-06:
Would be great as an extra MIDI FX. 
Although current behavior is just the nature of trackers

BlakRainbow, 2021-02-09:
I wouldn't want to have to program the note length for every single note like FX, but if it were integrated into the instrument parameters section it would be great. When midi is selected as an instrument, instead of the adsr section for a sampled instrument there could be extra midi options like note length.

Andreuparri, 2021-02-10:
Or it could be done with automatic note off like described in the following post
https://github.com/polyend/TrackerIssues/issues/779#issue-802795685

ConnorCarbon, 2021-02-21:
> Or it could be done with automatic note off like described in the following post
> [#779 (comment)](https://github.com/polyend/TrackerIssues/issues/779#issue-802795685)

I like this option a lot. 

But as a note to OP @BlakRainbow I know this is an unfair suggestion but I've been using a Launchpad Pro to trigger live sequencing and it sends OFF messages perfectly in alignment with human performance and I'm not trying to say "buy something new to solve your issue" but if you find yourself wanting OFF messages for a lot of stuff you want to live-sequence, I'd recommend getting some kind of controller that sends those messages, like the Launchpad Pro, etc. It works amazingly and is faithful to your performance

closed 2021-12-02 by Jacknametrouble