polyend/TrackerBetaTesting #777: [1.3.0] hanging MIDI notes

hubparticipant, 2021-02-06:
This is technically a feature request although I’m interpreting it as a bug. 

Currently when the sequencer is stopped or a MIDI track is muted, the Tracker doesn’t seem to send a note off over MIDI which causes hanging notes on external MIDI synths/devices. 

This is easily solved by having it send a note off message over all MIDI channels when the seq is stopped / tracks are muted by the user.

eholzbach, 2021-02-06:
I also want this, but only as a feature. It affects using the tracker as a master clock with downstream sequencers. You wouldn't be able to stop the tracker and have the downstream sequencers continue. Being able to toggle this on/off would fit more case uses.

ambv, 2021-02-28:
This happens when there's a sample following a MIDI instrument on a track without an explicit NOTE OFF.  The fix is to send a NOTE OFF to the relevant MIDI channel when a sample starts playing. This is symmetrical to all other event types where that already happens:
- MIDI instrument 02 playing after MIDI instrument 01 sends an implicit NOTE OFF to Channel 1
- MIDI instrument playing after a sample sends OFF to the sample
- sample 02 after sample 01 sends OFF to sample 01
- sample 01 after sample 01 sends OFF before playing sample 01 again
- MIDI instrument 01 after MIDI instrument 01 sends OFF before playing MIDI instrument 01 again

Also, muting a track should send NOTE OFF to the actively played MIDI note.

closed 2021-02-28 by ambv