polyend/TrackerBetaTesting #780: Left - and right arrow keys select next - previous slice

hubparticipant, 2021-02-07:
Chopping breaks is a bit tedious because of how you need to switch between slice selector and position. (F1 and F2 keys). 

Improvement:

While cursor is on “Position”, left en right arrow keys select next en previous slice in the sample editor - beat slicer. 

🙏🏻

danapeks, 2021-02-07:
Another potential option that comes up a lot is here - https://github.com/polyend/TrackerIssues/issues/217

hubparticipant, 2021-02-08:
Yes also. Slice follows MIDI input / MIDI input selects slice. 
Like on TAL Sampler, Logic, Maschine, pretty much every MPC since the MPC3000. I think every sampler in town has this feature. It’s a bit weird to see it isn’t on the Tracker