polyend/TrackerBetaTesting #782: Beat Slice - Pushing pad allows slice marker adjust

Supernatural802, 2021-02-07:
currently when you choose auto slice it lays the slice markers out on the pads and when you push a pad it plays from that slices marker start point. to adjust the marker start point you have to go to "slice" chose the slice number and then push "adjust" to be able to move the slice start point with the dial. when you trigger the slices by pushing the pads it jumps to whatever slice you are triggering on the pad and if you have "adjust" selected it still only moves the start point on the first slice regardless of what pad you are pushing to trigger the slice. Its not until you select the slice by using the "slice" button AND then pushing the "adjust" button can you move the slice start point marker. The other option that should be availabe , and better option in my opinion, would be to push a pad and trigger slice and at the same time have that be the slice that is now active for adjusting via the dial. it would make sense for a UX perspective. i hope this makes sense and we see this in a future update.

danapeks, 2021-02-07:
This has been raised several times - https://github.com/polyend/TrackerIssues/issues/217
Be great to see for sure, makes way more sense from a UX perspective, perhaps there's a reason why it's not been implemented that we don't know about. Be great to get a response from Polyend - they are pretty good at doing these sort of logical updates so i wonder if there's a reason?

BenisonBT, 2021-02-11:
I've thought that this would be the preferred way as well. Would make slice adjustment much faster.