polyend/TrackerBetaTesting #783: v1.3.0 - Instrument effect page UI bug

Supernatural802, 2021-02-07:
## Hi, sorry you had trouble!

### Steps to reproduce

push the Instrument button until you get to page 1, where you can pick effect. push effect button. you are on the instrument effect page - overdrive,bit depth, reverb send, delay send. push the Instrument button. push the instrument button again to get back to the effects page. When you return to the effects page overdrive and bit-depth are both selected in the red box and the names are bugged. Rotating the dial at this point affects the reverb amount. Push the back button and when you return the error is cleared. can reproduce every time. 

### Attachments and links

Video of bug - 
https://drive.google.com/file/d/1OPnF9phqxgLeeN6fRoZ_AlioZctHGwuo/view?usp=sharing

### Other important information

* How often does this bug happen? (every time, often, "it only happened once")
- can reproduce it every time. 

* Does this bug happen in previous firmware versions?
- not sure have not rolled back.

* Does this bug happen in every project?
- yes

MetabaronCZ, 2021-02-09:
Hi. This bug is related to https://github.com/polyend/TrackerIssues/issues/759. It should be repaired in latest beta as previously reported problem (https://github.com/polyend/TrackerIssues/issues/595) was closed.

Jacknametrouble, 2021-02-18:
Fixed in 1.3.1b1

closed 2021-02-18 by Jacknametrouble