polyend/TrackerBetaTesting #785: 'MONO' mode

talktogreg, 2021-02-08:
Consider the option for a 'MONO' for LIVE recording.

Explanation:
When entering notes (instruments/fxs), the Tracker is effectively working in held-MONO mode. By this, each note is held until it is replaced with the next. (you can cut it short by adding an OFF/CUT/FADE) or changing the ADSR envelope etc. The important bit is that the NEXT note cuts the last. Indeed you can also insert new notes in between existing ones, and the new will cut its predecessor and intern get cut by the next.

In LIVE rec mode though, we are essentially in 'POLY' mode, spread across the available Tracks as prescribed but the ARM settings. This is fine. 

But, if you only have 1 Track ARMed, the each note still places an OFF on release, and in fact you HAVE TO release the note before another is played. This is quite hard to do on the silicon pads. 

Could it hence be possibly to enter a more 'MONO' mode if only one Track is ARMed where if a new note is entered (silicon pads of via MIDI) it would cut the previous note (even if key is still being held). Additionally, this would allow for the Track to loop whilst recording and overdub.

My reasoning is for entering mostly bass-lines as a single Track. Having to carefully lift a key before the next is very difficult as hence notes get missed. The only other way around this is to ARM two or more tracks, and then copy back any notes that overflow, which is tedious.

------------
BONUS POINTS if 'last note priority' was supported. ie 
- play a pad (key) plays note
- whilst still holding the above, play another pad (key) - new note plays
- release 2nd pad (key), 1st note gets re-triggered. 

This is how MONO is supported on the majority of synths.
------------

tfspsound, 2021-02-11:
#132 and others

TwistedGenetics, 2021-03-19:
I so need this, it’s so frustrating overdubbing or just jamming live.

bambrose, 2021-03-19:
This is a great idea. Currently it is frustrating when trying to record when only one track is armed

ivanmalagon, 2021-07-19:
Yes please! This is extremely frustrating when entering bass lines or melodies.

UserOverflow, 2021-12-28:
Without overwriting notes in live mode, it's not possible to use tracker for live jam (jam - creating song live, not play already created song). I just installed 1.5b2 and I was trying to do that and worked quite well, but when I can't overwrite in live mode, then can't create in real live way.

ghartnett, 2023-09-25:
The new polyend backstage site has a wishlist with a feature like this. You can vote on it here. 
https://backstage.polyend.com/t/replace-notes-while-recording/3140