polyend/TrackerBetaTesting #787: Naming the cells in songmode

rotarycuff, 2021-02-08:
It would be great to be able to name the cells in the song mode, so it becomes easier to remember/see what is what when copy pasting. The tracknaming on the top is already something. But just being able to call them by number for example, would make the overall overview even better. (Even if it would only allow 2 or 3 digits to fit within the cell)

rotarycuff, 2021-02-08:
Posted in #535

ketumatek, 2021-02-08:
related post #421 and #535

Agree with you it will be easier for song mode if we could add colors and names...because just number it is for me to confusing...

LetsSeeIfWeSolveThis, 2021-02-08:
#363 I wrote this time ago, could be the same thing.

rotarycuff, 2021-02-08:
Ah yes indeed. Will lock this one then.

closed 2021-02-08 by rotarycuff