polyend/TrackerBetaTesting #788: [1.3.1 beta 3] can’t input notes via MIDI or USB

ToddTelfer, 2021-02-08:
## Hi, sorry you had trouble!

I am trying to use an external keyboard to play notes on the Tracker.
I have the CONFIG SETTINGS set to:

USB + MIDI In Jack
Channel 1 in video, but have tried all the channels including “ALL Channels”

I have tried both MIDI IN from a keyboard (Korg MicroKorg or Korg MicroStation or an MAudio Axiom) (also
tried from a Digitakt) and USB IN from an iPad and both Microstation
and Digitakt through the iPad AUM app.  I have tried all the channels
and tried setting the CONFIG channels appropriately instead of to “ALL”.

When I try to change the NOTES IN to either MIDI In Jack or USB, the
Tracker locks up, and must be unplugged to reset it.

FYI, the cable provided works great for MIDI OUT from the tracker to other devices.

Any help is appreciated!

### Steps to reproduce

try to input notes from a MIDI device or USB MIDI through an iPad or Mac.  

Try to change Input Notes settings in CONFIG.

### Attachments and links

Demonstrated here...

https://youtu.be/AwfcRtrfVzs

I’ve also tried this with RECORD engaged, but also it does not work. 

### Other important information

* How often does this bug happen? 

Every time.  The lockup happens even with nothing plugged in the MIDI ports when I try to adjust the settings.

* Does this bug happen in previous firmware versions?

I have tried 1.3.0 and this one.  Both have identical issue.

* Does this bug happen in every project?

Yes.  Both new and existing/example projects

ambv, 2021-02-26:
Did you input the notes with the REC mode enabled?

ToddTelfer, 2021-02-26:
> Łukasz

Yes, I tried it both in REC mode and in play mode.  FYI, someone saw my YouTube video and claimed to have fixed it on his machine...

He says he “reset the config in the firmware page” and now his Tracker no longer has the problem.  I can not test that fix because I no longer own the Tracker.

Thanks,

Todd

closed 2021-02-28 by ambv