polyend/TrackerBetaTesting #790: [1.3.1b3] Panning LFO sound artifacts

MetabaronCZ, 2021-02-09:
### Steps to reproduce

When using Panning LFO I hear periodic popping sounds (see attachment project). When I change tempo, these sounds goes faster or slower along with it. Which means the panning LFO changes pan value each step instead of gradually.

### Attachments and links

[panning-bug.zip](https://github.com/polyend/TrackerIssues/files/5954044/panning-bug.zip)

### Other important information

* How often does this bug happen? Every time

* Does this bug happen in previous firmware versions? Yes

* Does this bug happen in every project? Yes

doelman, 2021-02-09:
Yeah very annoying, already happened in previous firmware versions as well. Think I mentioned in a topic on here earlier. Makes the panning LFO quite unusable

x2db, 2021-04-13:
This mostly happen on low bass instruments (with a long amplitude)  When your instruments is pitched higher than 500Hz and more the artifacts are not as bad, Also while on this the random LFO could get some smoothing that would help with the pops and clicks when using random LFO panning

bambrose, 2021-04-13:
Yeah I've noticed definite clicking when random is assigned to panning. Is a shame cos it's a great effect

gordwithers, 2021-05-01:
Same problem for me. I also notice a lot of sound artifacts just adjusting the volume, whether in the step FX1/FX2, or even in the main mixer (which I've submitted in a separate bug report)

xelapo, 2021-07-11:
Same popping sounds here. It happens not only using Panning LFO, but just using several successive hard panning FX from one side to another.

OmskInfo, 2021-07-16:
Yep, just noticed this last night. 1.4.0 beta 3.

doelman, 2021-09-17:
@wjakobczyk-polyend @Jacknametrouble  is this on the agenda? It's quite annoying that LFO types, square and rev-saw are basically unusable because they often results in annoying popping and clicking sounds. There must be a way to avoid this, as other synths don't have this problem either.

bambrose, 2021-09-17:
It is frustrating as anything.
Renders the effect almost unusable for most melodic sounds... even plucks

ef780, 2022-01-22:
I'm using v1.5 and I am getting clicking with the random lfo set at 20% and above. Also there is clicking when using the ping pong loop mode. Makes its harder to employ a smooth sounding sample based synth instrument.

MortalJS, 2022-02-20:
Bug confirmed in 1.5.
@ziej8asek

closed 2023-04-21 by MiroPoly