polyend/TrackerBetaTesting #791: [1.3.1b3] Wavetable synthesis ADSR

MetabaronCZ, 2021-02-09:
### Steps to reproduce

I am using **Samples/Wavetables_2048/wt1.wav** from samples which goes with Polyend Tracker.

When I set it in Sample Loader as a wavetable, I see that the wave is **not** correctly justified (try yourself or see project in attachment).

Now run my project in attachment. When you play Pattern 1, you hear popping sound - there is no ADSR on instrument. When you go to instrument ADSR and set Attack to 0.010, half of popping is gone, which means the wavetable is really shifted and the start of single cycle waveform is not on 0-point. When you set release to 0.010 the sound is without popping. But when you set Release to 10 seconds the (end) popping is there again.

When you play Pattern 2 the popping cannot be cured even if you set Attack and Release to 0.010.

I know the popping on the note off is caused by cutting the wave cycle somewhere in the middle, but this could be fixed by rounding release curve (i mean to always drop note volume on end even when release curve is not finished).

I propose to set general ADSR (not just wavetables) minimum Release value to 1ms - 5ms. So the amplitude of sound would drop exponentially every time.

This 1ms - 5ms Release should be applied even when release curve is not finished and another note in Track will be played. 

I suppose your firmware is using something similar to WebAudioAPI, where audio node changes are scheduled, so there should be not problem with voice count - just exponentially droping volume 1ms - 5ms before note should end.

So there are 2 things in this ticket:
1) Waveforms in wavetable synthesis are probably not justified correctly - visually and sonically. Problem could be that tracker is corrupting wavetable file when loading / converting it into project?
2) A few milliseconds before any note ends, its volume envelope should go exponentially to zero.

### Attachments and links

[adsr-bug.zip](https://github.com/polyend/TrackerIssues/files/5954103/adsr-bug.zip)

### Other important information

* How often does this bug happen? Every time

* Does this bug happen in previous firmware versions? Yes

* Does this bug happen in every project? Yes

MortalJS, 2022-02-22:
Hi @MetabaronCZ 
Can you check if the problem occurs in firmware 1.5?

MetabaronCZ, 2022-02-24:
Hello. I just tried it and the problem is still there in 1.5.0.

It is the same as I described it in my first post in 2021. And the sample is still not correctly aligned in sample loader, which is probably the problem. I tried load the sample again in clean project, but it is the same.