polyend/TrackerBetaTesting #793: OFF message option

BlakRainbow, 2021-02-09:
ability to turn off the automatic OFF messages created by the tracker when live recording with an external midi keyboard.
The OFF messages end up muting half of the notes I'm trying to play on the keyboard.

talktogreg, 2021-02-09:
Hi. See my note re:MONO mode. I think we're on a similar page here.

BlakRainbow, 2021-02-09:
ah yes, definitely. I'm also having a hard time with held ON notes when sequencing external gear. 
It really sucks when you're trying to sequence a hardware synth with the tracker. Im used to being able to enter notes on a sequencer and have them trigger a 16th note (or whatever the grid is set to) while adjusting the adsr on my actual synth for note length...

Andreuparri, 2021-02-10:
Hi, I also asked for an automatic note off in here which would allow to set step length for new entered steps.
https://github.com/polyend/TrackerIssues/issues/779#issue-802795685

closed 2021-12-02 by Jacknametrouble