polyend/TrackerBetaTesting #799: Option in config menu to set default pattern length?

CliffRoberson, 2021-02-11:
On 1.3.0,  whenever adding new patterns from the song menu, unless I have edited them beforehand, they are set to length 32. This is a pain whenever I want a different pattern for each slot, because after I add the patterns, I have to go in and manually change the pattern length (either copy and paste or rotating the jog wheel for every pattern). Would be nice to be able to set the default pattern length (or maybe adjust all pattern lengths simultaneously).

sunfallz, 2021-02-12:
I 2nd this motion 👍

ConnorCarbon, 2021-02-21:
Yes! Simple but crucial. I am a 128-step pattern kind of person and I have been "fooled" by tiny little 32-step patterns many times after making new ones while in a creation flow. I'd personally love to just have any new patterns in a project match the previously-edited-pattern-length, so someone who made a 5/4 pattern that came out to be like, let's say, 80-steps, would have that same framework come up when creating a new pattern to keep their song going and matching the 5/4 metronome they hypothetically set.

But of course OP has a solution that would be "good enough" that I'd be fine with setting a default pattern length at the beginning of a project and going from there (but would love if it were global (cross-project) so I could set mine to 128 and call it good for most of my projects.

ConnorCarbon, 2021-02-21:
> (either copy and paste or rotating the jog wheel for every pattern).

Just a note to all that you can use the pads, too, for more quickly setting pattern lengths. Doesn't negate the need for this request but still will be good to know for people in the mean while