polyend/TrackerBetaTesting #801: [1.3.0] Tracker downgrading on its own

ROM72, 2021-02-12:
Hi,
I had a pretty big problem yesterday with the tracker :

I played a while with it then I turned it off. 
When I came back  to my office yesterday night, the tracker was turned on and had downgrade to firmware version 1.1, it was stuck at the 100 percent status of firmware update.
So I turned it off and on. 
It rebooted and I wanted to change the firmware version to what it was (1.3.0) but I don't know exactly what happened, my tracker shut down again and I counldn't turn it on anymore.
I feel very frustrated, why my machine is dowgrading on its own ? 
It shouldn't happen, It's so dangerous ! What if it happens on stage during a performance ? 

My tracker had the last stable firmware version which is 1.3.0 before it downgrade on its own.
I think I haven't lose anything because I can read what is on my sd card from my computer.

I managed to do a reset with the Polyend tool and my sd card is safe but still, this bug has to be fixed.  

### Steps to reproduce
I can't reproduce the steps. My tracker wasn't plug on a computer, 
but plug with its own power cord..

### Other important information

* How often does this bug happen? (every time, often, "it only happened once")
Once for now
* Does this bug happen in previous firmware versions?
I don't think so

jaap3, 2021-02-12:
I had something similar happen to me. Would be interesting to know if there are other ways to trigger a random downgrade, other than the hidden reset button next to the usb port. Could a power interruption also cause this?

thebradclay, 2021-02-12:
I think it has something to do with the the "randomly picked" part of this function. Something trigged that, I have no clue what but I bet this function is involved.
"press the hidden reset button on the back panel of the instrument and it will automatically flash with a randomly picked firmware .ptf file from the SD card Firmware/ folder.

Anntoin, 2021-02-16:
Encountered this behaviour once and it was quite surprising

srlowe, 2021-02-21:
I think I encountered this too.  Running v1.3.0.  I went to make a drink, and when I returned I was surprised to find that the screen was showing it updating to 1.2.1.ptf - stuck at 100%.   I left it for 10 mins, and after nothing changed I restarted it.  It booted up into v1.2.1, after which I upgraded again to 1.3.0.

thebradclay, 2021-02-21:
I only have one firmware installed at a time. I'm on the 1.3.1b3. After updating to whatever fw I want I then power off, remove the sd card, insert sd into laptop and remove all fw except the one I am using, that way if the tracker wants to change fw it can't lol

ROM72, 2021-03-12:
It happened again, I was jamming with my Machinedrum and Digitone, my Tracker was turned off. 
It turned on on its own and start downgrading again at version 1.1, and the update is caught in a loop, when it's at 100 it begin again from 0.
I am going to delete other firmware versions as a workaround but it would be nice to solve this issue.

I filmed the loop, at one moment I touch the tracker but I don't push any button.

https://www.dropbox.com/s/fali69y6rehokfr/tracker-loop-firmware.m4v?dl=0

Jacknametrouble, 2021-03-12:
Please get in touch with https://help.polyend.com/submit_ticket

closed 2021-03-12 by Jacknametrouble

ROM72, 2021-03-12:
I already have since February 12, Jacek from Polyend asked me to submit a bug report in github the first time it happened. 
Why do you close the issue if it is a bug ?

Anntoin, 2021-03-12:
I'm confused as to why this has been closed and the provided guidance. Should everyone that encounters this cut a support ticket? That doesn't seem right.

A number of people have encountered this and it is a valid bug so this issue should be re-opened

Jacknametrouble, 2021-03-12:
If you're facing the same issue, please get in touch with https://help.polyend.com/submit_ticket

thebradclay, 2021-03-12:
Yeah I'm with @Anntoin and @ROM72 on this. Why is this being closed?

Jacknametrouble, 2021-03-12:
Because this isn't a firmware bug.

thebradclay, 2021-03-12:
@Jacknametrouble okay, curious then what is causing it?

Jacknametrouble, 2021-03-12:
@thebradclay please get in touch with https://help.polyend.com/submit_ticket

thebradclay, 2021-03-12:
I just want the issue explained.

Jacknametrouble, 2021-03-12:
Good, just proceed as asked.

thebradclay, 2021-03-12:
done

ROM72, 2021-03-12:
I got a reply from Jacek from Polyend :
 
"It took us a long time to get to the conclusion with this issue.
This isn't a firmware bug. The part of the used reset buttons seems to be oversensitive. That's why the automatic downgrade happens from time to time.

We are able to fix this by reprogramming the unit's reset button itself. In order to do so, we need to get the unit that does so. This isn't anything user performable."

thebradclay, 2021-03-12:
@ROM72 I saw your comment before they deleted it. Thank you!

ROM72, 2021-03-12:
Wow, I don't know why they wanted to delete it, it's absurd. 
And I find it a bit brutal to close an issue without clearly explaining what's going to.

thebradclay, 2021-03-12:
wow

ROM72, 2021-03-12:
I just DMed you on fb Brad !

Jacknametrouble, 2021-03-12:
Anyone with such an issue please use https://help.polyend.com/submit_ticket. As you were already informed, this isn't a firmware issue. You were all informed on how to proceed with it.