polyend/TrackerBetaTesting #802: Pattern Chaining

ilambert, 2021-02-12:
I'd really like to see a quick way to chain patterns outside of song mode. 

It would be really nice to be able to hold down something like Shift+Pattern and tap a few squares in whichever order to have them loop.

tfspsound, 2021-02-13:
Made a similar suggestion #598 as a work around I don’t structure my song until it’s done and use Song mode to make a chain of patterns to work on.

sunfallz, 2021-02-13:
I second this "quick way to chain patterns outside of song mode" motion 👍

isbtn01, 2021-02-13:
Yes please. Would have been awesome to have this. Maybe could work similarly as on seq - putting an empty step to an end of a section that will redirect the actuve pattern to whatever pattern selected. On seq thats actually a genious idea.

tfspsound, 2021-02-13:
> I second this "quick way to chain patterns outside of song mode" motion 👍

Same idea as mine, ilambert has just added a way of doing it in Pattern mode, providing you can remember which patterns are on each pad you’d b fine. My patterns are rarely created sequentially and also number more than 48 sometimes so pattern 56 for example would be inaccessible using Shift + Pattern.

topherdavis, 2021-02-19:
https://github.com/polyend/TrackerIssues/issues/805 is this relevant/

closed 2021-12-02 by Jacknametrouble