polyend/TrackerBetaTesting #804: PADsynth

theorize999, 2021-02-13:
I commented about this in another issue but I really feel this deserves its own issue opened.

An extremely useful tool would be a sample generator that uses the PADsynth algorithm. This should be pretty easy to impliment, the algorithm has already been designed and is freely usable. If you do implement it please include some of the waveform generation options like the waveshaper and base function options, please see ZynAddSubFX to see a good implementation, the creator is who invented PADsynth. I've seen some other programs use it but they usually have an extremely simplified version that ends up being not as versatile, using PADsynth in ZynAddSubFX can render samples that are extremely varied and useful for so much more than pads. I've been using it off and on for quite a few years in many different contexts, if you want interface ideas or which features will matter most please contact me. One thing I will say is it would really help to have both discrete and bandwidth spetrum modes, discrete isn't quite single-cycle type of sound but it does sound quite a bit like a single oscillator but more analog than a single cycle. and the overtone controls also shape the sound a lot. 

Here is the address for the PADsynth algorithm:
https://zynaddsubfx.sourceforge.io/doc/PADsynth/PADsynth.htm

A VST version of ZynAddSubFX (old version that will export samples) so you can see the process, maybe export a couple samples and see how useful this would be with the tracker:
https://www.kvraudio.com/forum/viewtopic.php?t=268277

ZynAddSubFX Website that includes source code:
https://zynaddsubfx.sourceforge.io/

Also another idea, for single cycle generator would be something like what is found in Klystrack:
https://kometbomb.github.io/klystrack/
It is very easy to use with few controls, something similar would probably be very easy to get along with Polyend Tracker's controls. 
Actually I think the PADsynth could be implemented with relatively few controls as long as they were the right controls. For instance you can get away with only 16 bands for the additive portion, I've seen it that way and it works well, could maybe go as low as 8 or 12 but I'm not sure about that. But really all you would need is the base wave, additive controls, a waveshaper at that stage would be nice, they've often been useful to me, the filtering controls aren't really necessary, {but the more like the original the better, but not all options have a huge impact), also like I said earlier the spectrum modes and overtone controls have a huge impact on the sound... if you included the parameters above it would make a VERY powerful sample generator that could provide the tracker's sampler with plenty of material.

tfspsound, 2021-02-14:
Nice idea, makes it easier to write tracks that will utilise external gear that isn’t connected at the time of composition too.

cclcng, 2021-02-18:
This is an amazing thing !!!

closed 2021-12-02 by Jacknametrouble