polyend/TrackerBetaTesting #805: Request. - Song Mode modification -

ketumatek, 2021-02-13:
Related post #418 #421 #497 #518 #568 #571 #598 #753 #763 #781 #789
There are a lot of suggestions made in those post and I think put them all in one will helps.
I hope I will going forward with my request.

In song mode we have 8 tracks - Patterns 1 2 3 etc and slot 1 2 3 etc...
The 8 Tracks are connect to a pattern and if you want change some in a pattern you are making a new track.and it work in the same way for slot...

I will take 2 examples to be clearer of what I would like 
Renoise
[image]

Ableton live 
[image]

In this 2 examples we have tracks and in each tracks we can put a pattern select it - copy it - change it - put it in an other tracks - 
And as the polyend you can play the slot you are in.

1 request - would it be possible to think the tracker as Renoise or Live in that way in song mode

We have 8 tracks and we can copy paiste any pattern in any tracks and slot.

For example in the slot 1 I have all my tracks were there is information, and I want to copy in the slot 2 only tracks 3 and 4, and in slot 3 the tracks 1, 3 4 and 7 etc.

2 request - as in Live or Renoise in the slot section you can rename slot and the number don't change ...each time you put a new slot it is a new number. 

For the moment in song mode slot can be 
 1 5 6 3 16 5 etc 
And it not really easy when you want to change just few thing in a pattern.
In my ideas It will be always
1 2 3  4 5 etc.  Because you can copy and paiste each pattern from each tracks.

Exemple: I make a pattern call pattern 2
There 5 tracks in, and I am in slot 4 and in my song I have slot 5 6 7 where I already put something.
I want to make a modulation on my bass (tracks 3).
Actually I need to make a new pattern (4) copy my tracks in make my modulation for my bass and put this in a new slot (8), and put my new slot (8) after 4 and before 5 ...the result in slot session is 4 8 5 6 7 ...so more you have to make modulation more you have new pattern and new slot and after a moment it can be very confusing.
And if you just have to copy and paiste it is easier and quicker.

3 request - making loop with more than one slot 

Have the possibility to select 2 or more slot and loop them.

4 request - give some colors to the 8 patterns and the blocks inside the pattern as you can see in the 2 pictures.

I hope I don't forget stuff and it is enough clear...
Maybe the same ideas are in the relate post I put in the beginning...

maybe some stuff are not possible but if you can think about how song mode it is make in Renoise and Live and you can do something like that it would be amazing.

I know you're working hard on this machine and appreciate that.❤️

For the other please if you have more ideas writ them😉😉

tfspsound, 2021-02-13:
Thumbs up for the commitment alone to looking through all those posts on here.

tfspsound, 2021-02-14:
You put plenty of work into this one that’s for sure, good stuff. 
Not sure if I properly understand all of them though. 
1: I thought this was the case already since I thought you could set a blank pattern in a slot then Copy and Paste the tracks you want from one pattern to this new pattern. I remember assigning a blank pattern to a slot once but it may have been a pattern I had cleared data on, not sure, I use Copy Pattern in Pattern mode for this but doing it in Song mode could be quicker. Is this what you meant?
2: Hmmmm not sure I get this one. Slots and Patterns will align at first but will always become unlinked as your song progresses as you reuse Patterns right?
3: Yes. Just yes. As I mentioned elsewhere not structuring my songs until I have finished all my patterns so I can use Song mode to loop patterns is a pita work around not a solution, some sort of Play Selection is needed. 
4: Why not, personally I’d be more interested in colouring sections like intro or verse not individual tracks because they can be quite mixed up at times with the same sample across many tracks in the same pattern. 

Nice one

ketumatek, 2021-02-14:
> You put plenty of work into this one that’s for sure, good stuff.
>
> Not sure if I properly understand all of them though.
>
> 1: I thought this was the case already since I thought you could set a [...]

Thank for your answer.

1 - I love the facility we can copy and paste  a pattern really easy and very quick...in pattern mode.
And I would like to do the same in song mode.
The problem here is that you put a complete pattern in a slot and I would like to re-arrange this pattern more easily in song mode.
Change any blocks in a slot copy and paste a block in the track you want and in the slot you want...
The problem is that a slot is attach to a pattern ...
I think for this request they need to rebuild completely the mode song and may be the pattern mode too ...so maybe impossible..

2 - not easy to explane but if you have Live or Renoise I think it is easier to understand  what I mean.
But for me rename slot will help went you have a lot of slots to be clearer...same as color...

4 - it depends how you work ...but in general and for your final mix you will put your synth in a track your bass in a track your drums in a track etc...same as if you playing different sample of each one...
So colors will help to know what is were ...

tfspsound, 2021-02-14:
Cool. I’m sure 1 is possible now [1.3.0] isn’t it? I’ve only ever copied a track (blocks are tracks right?) to another pattern with the same number of steps though.

prds100, 2021-02-14:
Hey

I think this is a great idea, however I think the implementation is quite a big ask.  We are essentially asking Polyend to make Ableton in a box

I made a request for track mute automation in the song mode which i think is a more realistic request since then you are essentially creating new 'sections' of the song without creating new patterns or changing the fundamental nature of the sequencer. This would be as simple as a 'more' section in song mode where you can select to mute the track on the slot.  Currently we can delete tracks on slots but that affects the entire pattern and all slots where the pattern is placed in the song mode.

tfspsound, 2021-02-14:
Ahhh I get you now. Maybe they could insert a new column next to each track that was specifically a mute channel that is toggled on or off?

ketumatek, 2021-02-15:
> Ahhh I get you now. Maybe they could insert a new column next to each track that was specifically a mute channel that is toggled on or off?

Not necessarily you can in Song mode add a "mute section" or with "Shift mute" : select the block to mute and mute it or them by pushing the bouton Mute.
When mute is "on" there is a cross on the block muted.
As you can see in the first picture in Renoise.

topherdavis, 2021-02-19:
Just another example of chaining patterns

[image]

ConnorCarbon, 2021-02-21:
I'm not sure how feasible all of this is and since it's common for users to have sounds from many different instruments on a track, it's less intuitive (but certainly not counter-intuitive) to make room for pattern copy-pasting features if there's not much room, but if it's an easy add then sure! Would be great!

BUT! **Adding a SUB-FEATURE REQUEST to this post:**

- [ ] Request for deletion of a "cell" (meaning one track of one pattern) in Song Mode view will only delete that "cell" from that "version" of the pattern, and not all versions of that pattern on the Song Mode screen.

So for example if you Pattern 1 and copy it 4 times in song mode, and want to have the same pattern play but add 1 track each time, you could just copy the whole pattern 4 times as mentioned, toggle REC to access edit mode in Song Mode, delete the "cells" of the parts you don't want to play during this build-up. 

As it stands, trying this would delete that given track part on every instance of the given pattern (and possibly the pattern itself? I didn't check on that, I just kept undoing this because i was testing to see how it would behave if trying to remove parts from a pattern in song mode.)

prds100, 2021-02-21:
What you are describing is mute automation which is what I mentioned before.

dimatura, 2021-04-07:
My vote would be to allow each track in a slot to come from a different pattern (or be muted). Given that we're posting examples,
here's how this looks in buzz. 
[image]
The buzz sequence view is a bit more flexible, in that it shows the length of each pattern (and also apply some pattern transformations), but I could live without that.

headtopmusic, 2021-05-21:
Totally agree with all of these comments. The song mode is way too rigid. It would be awesome to be able to **mix and match** **patterns** at a **Track** level