polyend/TrackerBetaTesting #807: Enhancement: Shift+Wheel for moving cursor left/right on Pattern View

ConnorCarbon, 2021-02-14:
### Shift+Wheel Turning for allowing left/right cursor movement without the hand leaving the wheel

This one is pretty simple and I have noticed that this control combo is currently not assigned to a function, at least not in the Pattern view. The knob is great for scrolling up and down a track but to go left or right, it's all about repositioning the hand off of the knob and onto arrow keys, etc.

The idea here would be to allow holding SHIFT+Turning Knob left or right to move the cursor left/right (at probably a less rapid rate than up/down's rate / acceleration rate). For users with average right hands, this allows the right index finger to rest very comfortably on SHIFT while the rest of the hand turns the knob as normal, and the difference between scrolling up/down with the knob and left/right with it would then be only a press/release difference on the SHIFT key!

That's about it, other than noting this shortcut could be global to the tracker wherever scrolling left/right was applicable, on present and possible future menus that could benefit from users being conditioned for this method of input.

It's not dire, because of course the arrow keys can do this, but the repetitious repositioning of the hand back and forth between the keys and the knob eats up probably 1-2 minutes of time every hour session of use. After mastery (1,000 hours of use, let's call it), that might waste 16-33 hours of each user's time toward mastery on repositioning their hand to move the cursor a different direction and then resume knob control. If the user's professional "Work time" is worth $20/hr, that'd make adding this feature to the Tracker cause it to be $332+ more valuable to each user by the time they've mastered it.

If there are other plans for using up this shortcut that could fill a need that is otherwise not available on the Tracker through other control methods, I wouldn't worry about it, but if that shortcut mapping doesn't have any big plans, then I think adding this would be a huge time-saver, add to fluidity of control and user experience, and add value to each Tracker for each user.

tfspsound, 2021-02-14:
Personally I was hoping Shift + Jogwheel would increase the speed at which we could scroll through a waveform in Sample Playback and Sample Editor. That takes up far more of my time than moving left or right across the tracks. I agree though that going from Track 1 to Track 8 could be done quicker by screen wrapping the left and right like they do top to bottom. 
As there are only 48 sample slots I find it useful to use lots of sliced samples, I’d prefer to have a Jogwheel that lives longer than getting back very small amounts of lost time. I know there is the zoom function and I’m gradually getting used to jumping between Zoom and Adjust so my Shift + Jogwheel suggestion may also be redundant anyway. 
Not sure if holding Shift and turning Jogwheel on Master screen would be a quicker way to go through Filter and Fine Tune etc when there is a Screen/F key already assigned to that job.
Another quick point, if you were getting paid by the hour to make music wouldn’t it be in your interest to have the process take longer? You’d be out of pocket by at least $320 right?
Any way just adding my thoughts on this for discussion, I think we both agree there should be an alternative way of getting from Track 1 - 8 quicker and hitting Insert or Backspace by accident is a nightmare I’m so glad we have undo. 
Peace

ConnorCarbon, 2021-02-21:
> Personally I was hoping Shift + Jogwheel would increase the speed at which we could scroll through a waveform in Sample Playback and Sample Editor.

That is absolute necessity! But I think the Knob's acceleration functionality should apply to scrolling (and start and end-point movement, etc.) in the sample editing menu(s). It shouldn't even be a shortcut, unless the Knob's "rate-acceleration" can't be put to use on that screen.

You should search or there are already open issues about that and if not, make one, because it's a pet peeve of mine too. It is excruciatingly slow to move across a sample. Zoom can help, but not nearly as much as increasing the "scrub speed" as a video editor would call it.

closed 2021-12-02 by Jacknametrouble